prusa3d/PrusaSlicer #11089: Prusa 2.6.0 - No sparse layers adds a G1 Z.4 before executing the tool change g-code, smashing into the print

BogdanPNT, 2023-08-08:
### Description of the bug

Doing a "Change extruder" command while having a wipe tower enabled with no sparse layers enabled introduces a "G1 Z.4" in absolute coordonates into the gcode, smashing the nozzle into the print. 
With sparse layers enabled
[image]
With sparse layers disabled
[image]

### Project file & How to reproduce

To check the issue on the included project file, import it, slice it and go to 9.80 mm, layer 49. The problematic G1 Z.4 can be seen at line 4492.

To reproduce this issue, make sure to have a printer with 2 extruders selected, the wipe tower and no sparse layers enabled. Slide a model and do an extruder change, but not before 0.80 mm, because it seems like if you do an extruder change before that, the G1 Z.4 is not generated.
[SparseLayersIssue.zip](https://github.com/prusa3d/PrusaSlicer/files/12290412/SparseLayersIssue.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0+win64

### Operating system

Windows 10 Pro

### Printer model

Ender 3 running Klipper v0.11.0-257-ged66982b

capi92, 2023-08-22:
Same issue here.

Thus, as of now, we have 2 bugs preventing multi color print on MK4
MK4 firmware issue in M600 creating a blob :
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3106
This bug that is preventing wipe tower to work thus no way to remove the blob...

Please solve this asap...

BogdanPNT, 2023-08-27:
> @BogdanPNT you should rename the bug and upgrade description as per your finding
I've updated the title and description as per your request.

rkindl, 2023-08-28:
I have exact same issue with Prusa 2.6.0. And there is also a problem that after executing toolchange printing head is at first returned to model and only then it moves to wipe tower leaving some some small piece of filament on model and thus completly defeating purpose of WipeTower.

christianh17, 2023-09-03:
Same to me, I helped myself with editing the gcode before uploading but I hope for a solution in the slicer. My way is only reasonable for one or two changes, but I do not want to imagine what happens with many changes.

LouDnl, 2023-10-20:
Same issue here with Pruse 2.6.1
[image]
Just after CP TOOLCHANGE UNLOAD the printhead moves down into the print before moving to the wipe tower.
(ignore the red lines, they are comments in after toolchange custom gcode and unused)
[image]

Jan-Soustruznik, 2023-10-23:
Hello, 
@rkindl , @christianh17 , @LouDnl , could you share you 3MF file, please? 

Thank you, have a nice day

rkindl, 2023-10-23:
My model with correpsonding 3MF file is published here:
https://www.printables.com/model/404906-ark-nova-coffee-break-token/comments
In comments there is also a photo of the problem in version 2.6.0.

LouDnl, 2023-10-29:
Sorry for the wait, didn't have a 3mf file at hand. Recreated one, see attached zip file (couldn't upload the 3mf file).
It happens at this point on line 5516, layer 29 just after layer change.
[image]
[CHEPCalibrationCube.zip](https://github.com/prusa3d/PrusaSlicer/files/13198050/CHEPCalibrationCube.zip)

lukasmatena, 2023-11-11:
Fixed in [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). Closing.

closed 2023-11-11 by lukasmatena