prusa3d/PrusaSlicer #11091: "Detect bridging perimeters" options does the opposite

Niproblema, 2023-08-08:
### Description of the bug

As title says, this option does the opposite from what it says it does and ruins prints.

Detect bridging perimeters: DISABLED
[image]

Notice that feature type is correctly detected. Both fan speed and move speed are correctly applied for bridges.

Detect bridging perimeters: ENABLED
[image]
In this case bridges are marked as overhand perimeters. While correct fan speed and move speed are applied the move patern is completely wrong, since it assumes it can overhang layer lines mid air.. **This is very detrimental to bridging quality.**

Please find the attached model I used for testing bridge slicing.

### Project file & How to reproduce

[wing_bridge.zip](https://github.com/prusa3d/PrusaSlicer/files/12294285/wing_bridge.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 10 64bit

### Printer model

Voron 2.4 with afterburner

Jan-Soustruznik, 2023-08-09:
Hello, @Niproblema , 

I think that you have also enabled the new experiment feature implement in 2.6.0 > "Extra perimeters on overhangs" which detect overhang areas where bridges cannot be anchored, and fill them with extra perimeters paths. These paths are anchored to the nearby non-overhang area when possible. This feature cant applied for all cases, but in most cases can perform better surface and print stability.

Anyway, in filament setting > Cooling is now implement Dynamic Fan speeds where you can additionally control fan speed according to overlap.

Do you have any additional questions? 

Thank you, have a nice day

Niproblema, 2023-08-09:
Ah you are correct.