prusa3d/PrusaSlicer #11097: Organic support shifts into mid-air

chylld, 2023-08-10:
### Description of the bug

In some circumstances, organic supports that start under the object can't "get out of the way" fast enough and suddenly shift into mid-air. Using the current organic support settings as of August 10, 2023.

### Project file & How to reproduce

[Helmet organic supports.zip](https://github.com/prusa3d/PrusaSlicer/files/12307975/Helmet.organic.supports.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0+win64

### Operating system

Windows 11

### Printer model

Prusa XL

neophyl, 2023-08-10:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/10725

Jan-Soustruznik, 2023-08-10:
Yes, issue is duplicate of 10725. We working on it. I closed issue as duplicate.

closed 2023-08-10 by Jan-Soustruznik

xrbeattx, 2024-05-02:
Bump, its been almost a year, trying to print the Prusa Research rocket nozzle on a new printer (creality cr-m4)  and the supports are all f**ked up. I look at prusaslicer and notice that its actually printing what its being told and its the slicer thats messed up
[image]
[image]
Moving the part around and re slicing does not seem to solve it ... dissapointing