prusa3d/PrusaSlicer #11100: start_gcode Parsing error

Hr46ph, 2023-08-10:
### Description of the bug

Since early betas Prusa Slicer 2.6.0 with firmware set to Klipper throws an error during slicing:

```
G-code export to /var/folders/f5/tvcwd4052j54rpzxx38r_6yh0000gn/T/.18160.gcode failed due to invalid custom G-code sections:
start_gcode
Parsing error at line 1: Not a variable name
START_PRINT BED_TEMP={first_layer_bed_temperature[0]} EXTRUDER_TEMP={first_layer_temperature[initial_extruder]+extruder_temperature_offset[initial_extruder]}
                                                                                                               ^
Please inspect the file /var/folders/f5/tvcwd4052j54rpzxx38r_6yh0000gn/T/.18160.gcode.tmp for error messages enclosed between
        !!!!! Failed to process the custom G-code template ...
and
        !!!!! End of an error report for the custom G-code template ...
for all macro processing errors.
```

This start_gcode works with Superslicer and the variables exist in Klipper firmware macros.

I have seen another post about this during alpha or beta release but I can't find it anymore.

I assume its a bug, apologies if its something I can fix on my end I'd love to hear it.

### Project file & How to reproduce

Set firmware to Klipper
Add the following code to start G-Code:
```
START_PRINT BED_TEMP={first_layer_bed_temperature[0]} EXTRUDER_TEMP={first_layer_temperature[initial_extruder]+extruder_temperature_offset[initial_extruder]}
```
Obviously the referenced variable names need to exist in the macros you're calling.

Slice something.

Project file doesn't matter, its not a problem with an stl it happens with any file.
Screenshot is not included, the literal error is included above.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0+MacOS-arm64

### Operating system

macOS Ventura 13.4.1

### Printer model

Ender 3v2 with Klipper / MainsailOS on Raspberry pi 4

rtyr, 2023-08-10:
extruder_temperature_offset is not valid variable in PrusaSlicer (it is SS specific feature), so the error message is correct

Hr46ph, 2023-08-10:
Ah shoot! I didn't know that.

apologies, I'll close.

closed 2023-08-10 by Hr46ph