prusa3d/PrusaSlicer #11103: Excessive extrusion width in solid infill on inclined walls

Maiquess, 2023-08-10:
### Description of the bug

When there are inclined walls, the slicer performs solid infill on the inside of the perimeters. The problem is that the slicer generates print lines wider than those configured in "print settings > advanced".

In this field, we manually input, for example, an extrusion width of 0.4 mm. However, in the slicer, when there's an inclination, the extrusion width increases. This results in a loss of print quality when using 1-2 perimeters, as this excess filament leaves marks on the print

### Project file & How to reproduce

[image]
[image]
[image]
[image]
[config3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/12316481/config3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 11

### Printer model

all

jorge2014git, 2023-08-10:
Tenho o mesmo problema. Minha impressões ficam todas marcadas.

Jan-Soustruznik, 2023-08-11:
Hello, @Maiquess 

Thank you for your feedback. Since release 2.6.0 - alpha5(Link to changelog: https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha5)where was improved the ensure vertical shell thickness algorithm, is by default for solid infill used Arachne perimeter generator, also if classic perimeter generator is used in generally. So behaviour that you mention above is correct. About the quality lost... Are you sure, that the problem isnt caused any other printer settings or filament or printer? 

Have a nice day

closed 2026-09-29 by prusaslicer-bot[bot]