prusa3d/PrusaSlicer #11104: Organic supports have gaps in them

ProfessorPolygon, 2023-08-11:
### Description of the bug

When using Prusa Slicer 2.6.0 there are gaps in the organic supports. Seems to be when the angle of the supports change. Also the issue seems to get worse when I lower the layer height for example there are more gaps with a layer height of 0.12mm compared to 0.3mm. 

Seems this bug wasn't present in the Alpha6 release of Prusa Slicer, for me anyway.

[image]
[image]

### Project file & How to reproduce

[right sleeve back mould.zip](https://github.com/prusa3d/PrusaSlicer/files/12318075/right.sleeve.back.mould.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 10

### Printer model

Anycubic Kobra Max

neophyl, 2023-08-11:
Please do search for duplicate issues before posting.  This is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/11097 which was closed very recently as it was itself a duplicate of another https://github.com/prusa3d/PrusaSlicer/issues/10725

Jan-Soustruznik, 2023-08-11:
Yes, issue is duplicate of 10725. We working on it. I closed issue as duplicate.

closed 2023-08-11 by Jan-Soustruznik