prusa3d/PrusaSlicer #11105: FR: Favourite settings

ulab, 2023-08-11:
I wonder if we could have an option to "like" any of the Print or Filament settings, so they are displayed in a settings block on the right side of the Plater view.

This could replace the Supports, Infill and Brim settings that are displayed by default, since they could be "liked" instead.

The main idea is that I have a set of settings that I change more often than those. "Complete individual objects", "Support material / Style" and "Skirt loops" for example are more important to me. It would make things easier if I had them more accessible directly from the plater.

lars-sorensen, 2024-01-04:
This is really needed! So many clicks needed just to change settings to organic support.

ulab, 2024-01-04:
BTW: `Generate support material` and `Style` could be combined into a single setting `None, Grid, Snug, …`

lukasmatena, 2026-09-01:
Implemented in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena

ulab, 2026-09-01:
You've been waiting to close this, haven't you? :-)

lukasmatena, 2026-09-01:
Yes, like many others.