prusa3d/PrusaSlicer #11113: Parts of layer 1 missing after slicing

StunBmun, 2023-08-12:
### Description of the bug

Printing a jigsaw puzzle, about 200x167mm and 100+ pieces. Source is an AutoCAD drawing exported to STL.

Here is the 1st layer after slicing:
[image]

Here is the 2nd layer:
[image]

The number of pieces missing the 1st layer varies from 1 to about a dozen, located randomly within the puzzle.

The result is a very rough bottom surface to the affected puzzle pieces.

### Project file & How to reproduce

[Jigsaw 2023 B.zip](https://github.com/prusa3d/PrusaSlicer/files/12328219/Jigsaw.2023.B.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 11

### Printer model

Prusa MK3S+

sarusani, 2023-08-12:
There's some problem with [Elephant foot compensation](https://help.prusa3d.com/article/elephant-foot-compensation_114487). When you set it to 0, the first layer of your model will be sliced correctly.

JohnOCFII, 2023-08-13:
You might also benefit from improving the STL / reducing the number of errors.  

[image]

morphias2004, 2024-01-30:
Same issue and there are no errors in my models.

For prints that involve multi colour or multi material with chamfers directly off the build plate; is will generate perimeters in mid air as well, resulting in prints that fail in the first few layers.

The olnly solution is to set elephant foot compensation to 0 and use a deburring tool to clean up the finished parts.

J-Hoh, 2024-05-15:
I believe i encountered the same or atleast a related bug.
Here's the file im trying to print: https://www.thingiverse.com/thing:6379400/files the usb3cablecheckdeckelsicht.stl is the one causing trouble:

[image]
[image]
[image]

It's like its split into 2 objects, the lower plate and the tiny standouts letting the lid clip onto the actual casing. measuring in prusa slicer says those are apart by 0.002mm.
Printing in 0.15mm layers is fine, switching to the default 0.2mm preset causes a layer to just go missing 🤷

[image]

Hope this helps in debugging

neophyl, 2024-05-15:
J-Hoh your model is not the same issue at all.  The reason theres a gap when sliced is that there actually is a gap on the model.
[image]

If the layer height falls into the gap then you get a layer 'missed'.  It is not actually missing though as its there on the model.  Change the slice layer height to 0.1 for example and no more gap as  parameters of the slice are different.   Of course the best fix is to actually fix the model so that it doesn't have a gap in the geometry

closed 2026-09-29 by prusaslicer-bot[bot]