prusa3d/PrusaSlicer #11115: can you add more anycubic printer profiles

Bigfoot3387, 2023-08-13:
not one anycubic printer listed that are modern her is a list that needs to be added thanks
Anycubic Kobra 2
Anycubic Kobra Max
Anycubic Kobra Neo
Anycubic Kobra Go
Anycubic Vyper
Anycubic Kobra Plus
Anycubic Kobra
no profiles for any of these i want to use your slice and pritables but this is holding me back

neophyl, 2023-08-13:
Github for the slicer settings is here https://github.com/prusa3d/PrusaSlicer-settings

Also all non prusa profiles are user submitted, so until someone actually submits one then it cant be added.  If you have a printer that isn't on the list then you could create a profile , test it and then submit it and help other people out.

There are a few threads on the Prusa Forums PS section https://forum.prusa3d.com/forum/prusaslicer/  that have Anycubic profiles in them you could check out to start with.

Bigfoot3387, 2023-08-13:
Ok I didn't know first time I have tried prusha slicer and I'm already pulling my hair out trying to just get it working this is one of the most difficult slicers I've tried I'll just uninstall it and stay with cura but thanks for the info

closed 2023-08-13 by rtyr