prusa3d/PrusaSlicer #11117: 搭桥路线存在严重问题，各个区间之间存在不闭合的间隙，导致打印失败

lantinys, 2023-08-13:
### Description of the bug

[image]
[image]
[主体.zip](https://github.com/prusa3d/PrusaSlicer/files/12329620/default.zip)

### Project file & How to reproduce

直接切片就行

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

window10

### Printer model

启庞Kp3sPro

rcmaniac25, 2023-08-16:
Used Google translate... This may be a duplicate of #10533 as bridging doesn't actually attach in side walls.

ddacunha, 2023-09-12:
Seems related to #9999

lukasmatena, 2024-03-21:
Fixed in [PrusaSlicer 2.7.3-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-alpha1). Closing.

closed 2024-03-21 by lukasmatena