prusa3d/PrusaSlicer #11119: Measure tool disabled when there's an object "sinking"

sarf2k4, 2023-08-13:
### Description of the bug

Measure tool are disabled when there's another object is "sinking" below the bed platform

### Project file & How to reproduce

Just get any object "sinking"

[x-end-motor-R2.zip](https://github.com/prusa3d/PrusaSlicer/files/12330261/x-end-motor-R2.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0+win64

### Operating system

windows 10 64bit

### Printer model

prusa mk3

Zogar89, 2024-10-22:
I've run into this issue today, I wish it could be fixed.

pavolNovak, 2024-12-09:
I had the same issue today , version 2.8.1 , I somehow figured it out its the problem of sinking and then it was working.

mabblernet, 2025-01-26:
Same here - really annoying. Endet up having to cut the object that was partly below the print bed instead...

lukasmatena, 2026-09-01:
Fixed in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena