prusa3d/PrusaSlicer #11121: Supports are not from first layer

weed0wizard, 2023-08-14:
### Description of the bug

Hi. I have wierd situation with organic supports.
Using 0.2 mm layer height everything is normal, but all layers height below causing making supportst "in the air, from 14 layer in my case) I tried different settings but all the time the result is the same

### Project file & How to reproduce

[jaguar project.zip](https://github.com/prusa3d/PrusaSlicer/files/12333432/jaguar.project.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.6.0

### Operating system

Windows 10

### Printer model

Alfawise U20 on Marlin 2

neophyl, 2023-08-14:
Please search for duplicate issues before opening new ones.  This is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/9601

kubispe1, 2023-08-21:
It is duplicate, as was mentioned. Thanks

closed 2023-08-21 by kubispe1