prusa3d/PrusaSlicer #11123: Prusa Slicer 2.6 crashes every time I run but at random times

tomservaux, 2023-08-14:
### Description of the bug

Every single time I run Prusa Slicer MS error 0xc0000409. It doesn't seem to have a specific cause. It could run for 5 seconds and crash or it could crash when I load a file, open print settings, try to slice anything. It started after I installed version 2.6 release. I am ready to stop using it at this point because i can't get anything done and I do this for a living. One thing I noticed is that Bambu Slicer, which is a port of Prusa Slicer, also crashes at odd times, although not as often or consistently. I have attached the Windows Event View log for the latest crash.

### Project file & How to reproduce

1. Open Prusa Slicer
2. Do anything
3. Eventually it crashes

[Prusa Slicer Crash.zip](https://github.com/prusa3d/PrusaSlicer/files/12335967/Prusa.Slicer.Crash.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 + Win64

### Operating system

Windows 11 version 22621.2134

### Printer model

I've witnessed the crash on Prusa Mk i3 Mk3S+ and Voron 2.4.1

tomservaux, 2023-08-14:
Looking into this further it seems that the slicer usually crashes when arranging or slicing. As mentioned, I am having the same exact problem with Bambu slicer. I have filed a support ticket with them as well. 

One other thing, and I am not sure if this is related, when I restart after a crash, sometimes it comes up saying that this is the first time I have run Prusa Slicer. Just including this since it only happens after a crash.

tomservaux, 2023-08-14:
I tried 2.5.2 since that worked before and it also crashes when loading projects

Jan-Soustruznik, 2023-08-15:
Hello, @tomservaux , 

Did you try to check the HW on your PC. if problem is random and can be reproduced also in 2.5.2 which works before...  It may be a HW issue.

Anyway, I suggest try to fully re-install slicer / delete the configuration folder  (NOTE make backup of your custom profiles before re-install)

tomservaux, 2023-08-15:
Jan,

Thanks for getting back to me.

I thought of that because I did just replace the memory on my PC but I have done a full memory diagnostic and no issues were found.

Are there other components which might cause this issue? I run a lot of apps but I'm only having this crash with Prusa slicer and forks of it. Did the log files I sent have any info that point at hardware?

Best,

Chris

tomservaux, 2023-08-15:
I forgot to mention that I did a complete uninstall and reinstall of prusa slicer including deleting every single thing on my PC related to it, including my config files, to no avail.

Best,

Chris

closed 2026-09-29 by prusaslicer-bot[bot]