prusa3d/PrusaSlicer #11124: Missing part of a model after slicing

Maccraft123, 2023-08-14:
### Description of the bug

A part of a model(tail) is missing in g-code view, even though in preview it's all there. The issue happens regardless whether i set perimeter generator to "Arachne" or "Classic". Meshmixer displays the OBJ file just fine, STL also has the same issue. Attaching both files in a ZIP archive as Github didn't let me attach a bare STL or OBJ file.
Other models are sliced perfectly fine, only this one has been sliced wrong.

This is how it looks liike in model preview:
[image]

This is how it looks like after slicing:
[image]

[Model.zip](https://github.com/prusa3d/PrusaSlicer/files/12337788/Model.zip)

### Project file & How to reproduce

1. Open OBJ/STL file that was included with this Issue
2. Slice the file

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0+win64

### Operating system

Windows 10 64-bit

### Printer model

Heavily modified Creality Ender 3

jmd73, 2023-08-14:
@Maccraft123 There are errors in the STL. Repair it by clicking on the triangle by the object name.
[image]

Maccraft123, 2023-08-14:
Thanks, this worked