prusa3d/PrusaSlicer #11125: Slicer Creating Last Wipe Tower Move In Air Above Model

DevJim, 2023-08-14:
### Description of the bug

When the wipe tower doesn't have color changes all the way to the top of trhe model and stops building the tower, it finishes with a mid air wipe.

[image]

### Project file & How to reproduce

[Slicer Creates Wipe Tower Line in Mid Air.zip](https://github.com/prusa3d/PrusaSlicer/files/12338628/Slicer.Creates.Wipe.Tower.Line.in.Mid.Air.zip)

Steps to Reproduce:

Open attached project in zip file.
Hit "Slice now"
Viewport window of sliced objects shows floating lines above the wipe tower.

Or...

Create object that has color changes at the bottom of the of the print, but not at the top.
Add object to project.
Choose Slice now.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0+win64

### Operating system

Windows 10

### Printer model

Prusa MK3S+

neophyl, 2023-08-14:
Please search both open no closed issues before opening new ones. Duplicate of many issues here such as  https://github.com/prusa3d/PrusaSlicer/issues/4005  This behaviour is intended as mentioned in the linked issue apparently.

jmd73, 2023-08-15:
@DevJim As neophyl stated and it is in the [FAQ](https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)).

closed 2023-08-15 by Jan-Soustruznik