prusa3d/PrusaSlicer #11128: Support half height perimiters

bramcohen, 2023-08-15:
With variable width of filament laid down it should be possible to get the greater accuracy of a thin layer height at the perimeter of a model while still getting the speed of a thicker height on the interior. For each layer first print a half-height layer at the perimeters, then make a whole-height layer of the whole interior, then make a second half-height layer of the upper part of the perimeters. The trick is meshing together the two different perimeters with the same interior, but once that's worked out the technique should work great.

neophyl, 2023-08-15:
Of course you could just set your perimeters to the desired height and then use the already existing combine infill setting to generate your infill at double the height.

bramcohen, 2023-08-15:
> Of course you could just set your perimeters to the desired height and then use the already existing combine infill setting to generate your infill at double the height.

Thanks, I didn't know the feature was already there! I'll close this issue out (although it would be nice if it could double internal perimiters)

closed 2023-08-15 by bramcohen

neophyl, 2023-08-15:
Theres already a feature request somewhere for that around.  The double height internal perimeters I mean.