prusa3d/PrusaSlicer #11130: Travel blocker

jaggzh, 2023-08-15:
A modifier which prevents travel moves through it.
We came up with a couple use-cases in irc #reprap when trying to solve a problem:
1. Would allow inserts that protrude above the already-printed area
2. Would allow "support inserts", like let's say you put a soft silicone insert that could tolerate the head butting up against it, but not travel moves through it.
3. Knock-over reduction: Allow the user to prevent moves in a direction where they determine the part might have a high chance of being knocked over.

neophyl, 2023-08-15:
This was actually requested in Oct 2022.  Unfortunately they closed the issue as changes in 2.6 were deemed enough to 'fix' it.
https://github.com/prusa3d/PrusaSlicer/issues/9081

pbedenbaugh, 2023-10-03:
This is an excellent request. 
User interface should be similar to paint on supports, paint surfaces for no travel, or perhaps a modifier, with a solid shape defining a no-travel zone.

Stringing can be difficult to remove from tight places, and can prevent mating of parts, for example consider the Broadcom HFBR-4532Z fiber optic connector. It should have surfaces to align it over an extended range in order to properly align the beam. However remnants of stringing can block proper mating of the connector with a printed part. If the printed part dimensions are adjusted to insure a proper fit, you no longer know which dimensions are correct, so even if you can print one, you don't know the proper dimensions for outsourcing a production run.

yaderhey123, 2023-10-06:
Thank you for bringing this up again. I had requested it previously, but the post was closed even though I don't think anything was actually implemented. I would love to see this addressed. For me the stringing with TPU was the primary motivator, but I love the idea of allowing for mid-print inserts.

AJolly, 2024-03-29:
yeah this would be great

yuvs0, 2025-02-03:
Running into an issue with a part I'm printing and this would be a much quicker, simpler, and cheaper solution than spending time and filament dialing in oozing and retraction settings to ensure 0 stringing.

Gehetzter, 2025-03-02:
I would need this modifier too. I've got stringing always in the worst to remove places. Would be great to define areas that are avoided by travel moves.

Tupson444, 2026-09-12:
still relevant