prusa3d/PrusaSlicer #11132: Random positioning of initial purge line

m-cas, 2023-08-15:
**Is your feature request related to a problem? Please describe.**
Having the purge line at the same place managed to damage on the long run my original mk3 pei sheet.

**Describe the solution you'd like**
Select randomly where to start the purge line so that the whole front is used.

**Describe how it would work**
The G-code for positioning the line should have a random X component.

**Describe alternatives you've considered**
I've added a move to another location but this is not a dynamic one.

**Additional context**
no

m-cas, 2023-08-16:
Ok, I've decided that if I can managed to write operating system in C and assembler I've got to solve this and I did!
here is the working code I'd love to see making it into production.

;
; Extrude purge line
;
G92 E0 ; reset extruder position
G1 E{(filament_type[0] == "FLEX" ? 4 : 2)} F2400 ; deretraction after the initial one before nozzle cleaning

; Original Prusa code commented out
;G0 E7 X15 Z0.2 F500 ; purge
;G0 X25 E4 F500 ; purge
;G0 X35 E4 F650 ; purge
;G0 X45 E4 F800 ; purge
;G0 X{45 + 3} Z{0.05} F{8000} ; wipe, move close to the bed
;G0 X{45 + 3 * 2} Z0.2 F{8000} ; wipe, move quickly away from the bed

; Random start of purge line inside the first half of the plate
{
local rndpos = int(random(0, print_bed_max[0] / 2));
"M117 Purge begins at " + rndpos + "; share the new start
";
rndpos = rndpos + 15;
"G0 E7 X" + rndpos + " Z0.2 F500 ; purge
";
rndpos = rndpos + 10;
"G0 X" + rndpos + " E4 F500 ; purge
";
rndpos = rndpos + 10;
"G0 X" + rndpos + " E4 F650 ; purge
";
rndpos = rndpos + 10;
"G0 X" + rndpos + " E4 F800 ; purge
";
"G0 X" + (rndpos+3) + " Z{0.05} F{8000} ; wipe, move close to the bed
";
rndpos = rndpos + 3 * 2;
"G0 X" + rndpos + " Z0.2 F{8000} ; wipe, move quickly away from the bed
";
}

rtyr, 2023-10-04:
Thanks for your suggestion. We will discuss this internally, but I think we will not change the default behavior for now. The main reasons we changed previous behavior (purging close to the object) were ease of removal by hand without touching the bed and purging outside of print area. This change also needed some updates in the firmware.