prusa3d/PrusaSlicer #11136: Lock up (crash) on font selection

avxmw, 2023-08-16:
### Description of the bug

> What is the problem?

Trying to add/cut text via the tool introduced in 2.6, the application locks up or crashes when trying to select a font other than the default selection.

With the text element highlighted and trying to `edit text` I can change dimensions and orientations without problem, however opening the drop-down list of fonts in search for a different font makes the slicer stall for minutes (let it run for 20min+) at 100% CPU showing `Queue` behind every font and a spinning cursor. My font list is not really large and even if it would be, whatever the process does should be done in a reasonable time frame.

> What did I expect?

I expected to pick a font and use it like in any other application

[image]

Running the flatpak from CLI yields no interesting output, every possibly useful right has been given using `flatseal` compared to the default setup. Total list of fonts is 851 including lots of variants (NotoSans is default).

### Project file & How to reproduce

1. Import some .stl/.step
2. right click, select add part/negative volume or anything which lets you add text
3. try changing font

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 (flatpak)

### Operating system

openSUSE Tumbleweed Linux 6.4.9-1-default x86_64

### Printer model

Flashforge Creator Pro 2, Artillery Sidewinder X2

Jony01, 2023-08-29:
Could you please provide a text log from run of application?
To create one you need to start application with parameter `loglevel=5` from command line.
e.g. `./PrusaSlicer-2.6.1-rc1+linux-x64-GTK3-202308221615/prusa-slicer --loglevel=5`

I think that this issue is independent on model and could be reproduced by:
1. start application
2. hit the key 't' for open text tool
3. try to change font
(please confirm that issue appear by this steps)

In my opinion it will be some magic behind your operating system, which I am not familiar with. But we could handle it ;-)

avxmw, 2023-08-29:
Here you go, did as told, crashed while after clicking on a font different then default.

I get to see 3-4 font previews - which, sidenote, are rather useless as the font-window is tiiiiiny by default. When scrolling to the top of the font list everything has a status loading or queue, seems like previews for all fonts are requested in parallel and that is too much, but even if, this should be done after some time and certainly not the 20min+ I once let it sit on.

[prusa.log](https://github.com/prusa3d/PrusaSlicer/files/12461559/prusa.log)

Jony01, 2023-08-29:
There were same changes in font listing in new version(2.6.1) please try it in the latest version:
https://github.com/prusa3d/PrusaSlicer/releases

From the log it seems like fonts are loading in s couple of milisecond.
```
[2023-08-29 09:29:01.990861] [0x00007f4978bf5d40] [info]    Job 'N6Slic3r3GUI18CreateFontImageJobE' spend 20ms (process 20ms + finalize 0ms)
[2023-08-29 09:29:02.006066] [0x00007f4978bf5d40] [info]    Job 'N6Slic3r3GUI18CreateFontImageJobE' spend 15ms (process 15ms + finalize 0ms)
[2023-08-29 09:29:02.020470] [0x00007f4978bf5d40] [info]    Job 'N6Slic3r3GUI18CreateFontImageJobE' spend 14ms (process 14ms + finalize 0ms)
[2023-08-29 09:29:02.034954] [0x00007f4978bf5d40] [info]    Job 'N6Slic3r3GUI18CreateFontImageJobE' spend 14ms (process 14ms + finalize 0ms)
[2023-08-29 09:29:02.049469] [0x00007f4978bf5d40] [info]    Job 'N6Slic3r3GUI18CreateFontImageJobE' spend 14ms (process 14ms + finalize 0ms)
[2023-08-29 09:29:02.063946] [0x00007f4978bf5d40] [info]    Job 'N6Slic3r3GUI18CreateFontImageJobE' spend 14ms (process 14ms + finalize 0ms)
[2023-08-29 09:29:02.078709] [0x00007f4978bf5d40] [info]    Job 'N6Slic3r3GUI18CreateFontImageJobE' spend 14ms (process 14ms + finalize 0ms)
```

Could you also capture the video. I would like to see what happens on your OS?

BTW: Fonts are loaded parallel but with limited count of thread. Current proccessed on thread are labeled as `processing` and all additional are labeled as `queued`. Also the threads for fonts currently not visible are interupted. But still there are part of code which could not be interupted in the middle of proccessing (like providing font from OS)

avxmw, 2023-08-29:
There you go, recording and the log belonging to it.
https://github.com/prusa3d/PrusaSlicer/assets/97355652/a4269493-06f3-47c0-9e12-7e824b3ff71f

[prusa-new.log](https://github.com/prusa3d/PrusaSlicer/files/12463221/prusa-new.log)

closed 2026-09-29 by prusaslicer-bot[bot]