prusa3d/PrusaSlicer #11140: Add Setting - Extrusion Multiplier on objects on buildplate

kageryu42, 2023-08-16:
**Is your feature request related to a problem? Please describe.**
It would be nice to be able to assign specific extrusion multipliers to objects on a buildplate.  This would make for an easy calibration of the extrusion multiplier as one could see the results of different multipliers at the same time.  I think Super Slicer has a calibration print that does something similar.

**Describe the solution you'd like**
Right click an object on the plate, go to "Add Setting" and then select Extrusion Multiplier.

**Describe how it would work**
This should work like adding any other object specific setting like Extrusion Width, Fuzzy Skin, etc...

**Describe alternatives you've considered**
The alternative is to print out the object one at a time and then compare.  It would be much more efficient to compare multiple objects at the same time.

**Additional context**
N/A

rtyr, 2023-08-17:
I am closing this as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/10192.

closed 2023-08-17 by rtyr

kageryu42, 2023-08-17:
Tank you, this is actually better than my suggestion.  I hope it makes the cut.

 

Thanks,

Phil