prusa3d/PrusaSlicer #11143: No (visible) error for invalid 3mf file

fallingcats, 2023-08-19:
### Description of the bug

Trying to open an invalid .3mf file does not show any errors and does not launch PrusaSlicer.

After a couple minutes I figured out it would print an error to stdout when launched through a console window, but I think this deserves a proper error instead of failing silently for gui desktop users.

```
[2023-08-19 19:57:50.281687] [0x00007fc4de01a8c0] [error]   Found item with invalid object id
[2023-08-19 19:57:50.281707] [0x00007fc4de01a8c0] [error]   Error (Invalid 3MF format) while parsing '3D/3dmodel.model' at line 6305
[2023-08-19 19:57:50.281712] [0x00007fc4de01a8c0] [error]   Archive does not contain a valid model
export by FreeCAD.3mf: Loading of a model file failed.
```

I'm also not sure this file is actually corrupted, the file seems fine and Cura opens it without complaining. Please let me know if I should report this in a seperate bug.

### Project file & How to reproduce

[export by FreeCAD.zip](https://github.com/prusa3d/PrusaSlicer/files/12386456/export.by.FreeCAD.zip)

- Open the .3mf file with PrusaSlicer
- Nothing happens, no (visible) error 

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0+UNKNOWN

### Operating system

Arch Linux

### Printer model

Creality 3 max neo

Florin-Popescu, 2023-08-20:
Hello!

I'm getting the same issue, not on all models but it looks like on models with multiple parts in freecad or the ones where I took an stl file, imported and modifed it (which created several objects, shape binders etc. under the same part).

Looks like freecad changed the format of their .3mf exports in v0.21: FreeCAD/FreeCAD#10075 or FreeCAD/FreeCAD#9731.

> The problem is that with 0.20 all selected objects are converted into a single mesh and this single mesh is written to the 3MF file while with 0.21 several objects are written to several meshes which Prusa apparently doesn't like.
> 
> A workaround for now is to first create the meshes using the Mesh wb, merge them to a single mesh and then export it to 3MF.

Tested just now. A file exported with FreeCAD v0.21 doesn't open in PrusaSlicer with exactly the same error as yours, but a file exported with FreeCAD v0.20 opens correctly. If you're looking at the 3dmodel.model file in the .3mf, it just changed `type="model"` for `type="surface"` in the object's attributes. But that's for a single part exported. For multiple parts exported from FreeCAD I'm not sure if the 3dmodel file would have more changes.

Just hoping this gives the PrusaSlicer team more info to work with.

fallingcats, 2023-08-21:
> > The problem is that with 0.20 all selected objects are converted into a single mesh and this single mesh is written to the 3MF file while with 0.21 several objects are written to several meshes which Prusa apparently doesn't like.

I've only exported a single body and still have this issue

Luoti, 2023-09-30:
Would of course be cool if fixed, but anyone else finding this on google (like me). There is an option in the new version of Freecad  to "force to always write a mesh as model type even if it's not a solid" https://github.com/FreeCAD/FreeCAD/pull/10446 which should allow you to import the files to PrusaSlicer.

jaywgraves, 2024-02-19:
@Luoti  I did find this :tada:  The PR you linked to was backported and I'm not sure what release it's in but I upgraded from 0.21.0 to 0.21.2 and my troublesome FreeCAD 3mf exported part now loads fine in PrusaSlicer.

closed 2026-09-29 by prusaslicer-bot[bot]