prusa3d/PrusaSlicer #11144: Slicing issue resulting missing part of the piece

clawfire, 2023-08-19:
### Description of the bug

I'm importing my stl file. The preview looks good. I can pain my support without any problem and position the element as I like. But when I slice it, the circle part that should be plain are cutted on several millimètres with no print at all. It happen symetricaly on my piece, which is weird. 

### Screenshots 

[image]
[image]

### Info

PrusaSlicer
Version:   2.6.0+MacOS-arm64
Build:     PrusaSlicer-2.6.0+MacOS-arm64-202306191404

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Ventura Version 13.4.1 (c) (Build 22F770820d)
Total RAM size [MB]: 17,180MB
OpenGL installation
GL version:   4.1.0
Profile:      Core
Vendor:       Apple
Renderer:     Apple M1 Pro
GLSL version: 4.10.0
Textures compression:       Enabled
<details>
<summary>Installed extensions:</summary>
GL_APPLE_client_storage
GL_APPLE_container_object_shareable
GL_APPLE_flush_render
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_texture_range
GL_ARB_ES2_compatibility
GL_ARB_blend_func_extended
GL_ARB_draw_buffers_blend
GL_ARB_draw_indirect
GL_ARB_explicit_attrib_location
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_occlusion_query2
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_subroutine
GL_ARB_shading_language_include
GL_ARB_tessellation_shader
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_cube_map_array
GL_ARB_texture_gather
GL_ARB_texture_query_lod
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_swizzle
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_texture_compression_s3tc
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_sRGB_decode
GL_NV_texture_barrier
</details>

### Project file & How to reproduce

Just open the project and slice it. You should see a difference of rendering between the stl been just imported and the generated G-code rendering / sliced file.

[FFM v2.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/12386873/FFM.v2.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0+MacOS-arm64

### Operating system

macOS 13.4.1 (c) 

### Printer model

Original Prusa MK4

mix579, 2023-08-20:
The triangles in the parts list tell you that each of the parts has problems that should be fixed before trying to slice it. On Windows you can use the Fix with Netfabb option but that's not available on the Mac. Instead you may want to try https://formware.co/OnlineStlRepair which I found works very well.

clawfire, 2023-08-20:
Hey @mix579 , thanks for the reply. I tried to over the triangle but got no infos from it so I was a bit puzzled :(. Also, other stl with the same symbol appearing print just fine.

mix579, 2023-08-20:
Yeah, Sometimes PS can (silently) fix problems, sometimes it can't...

clawfire, 2023-08-20:
Tried the tool you mentioned and it can't fix my STL. I tried also the prusa slicer fix STL function (there's one which create a .obj file on macOS) but doesn't fix it either. 

Would be usefull to display a note on hover of the warning triangle though.

foreachthing, 2023-08-21:
You really should merge your mesh.
I don't know the software you used to "patch" this together, but Blender (3D-printing addon) complained about intersecting faces and so on.
Basically, you have twelve(!) separate parts mixed together - but they are not merged! Nodes don't touch each other. No wonder the slicer has problems with it. I'd be surprised if it didn't...

Options:
* **Merge** your mesh with boolean (CAD, Blender, etc)
* **Remesh** using other software (ftetwild, meshlab, ... (Ansys refused to remesh it))
* **fix** your model in CAD, where it came from

I'm running it currently through ftetwild ... but that takes a while to finish. I'll post it if it's any good.

foreachthing, 2023-08-21:
Please try this model, fresh out of ftetwild with these parameters:
`floattetwild_bin -i FFM_v20.stl -l 0.005 -e 0.0005 --stop-energy 9 --level 3 --smooth-open-boundary --manifold-surface --use-floodfill`
[FFM_v20.stl__sf.zip](https://github.com/prusa3d/PrusaSlicer/files/12394008/FFM_v20.stl__sf.zip)

foreachthing, 2023-08-25:
@clawfire did this help your case? If so, please close the issue.
 Thanks!

clawfire, 2023-08-26:
> Please try this model, fresh out of ftetwild with these parameters: `floattetwild_bin -i FFM_v2.stl -l 0.005 -e 0.0005 --stop-energy 9 --level 3 --smooth-open-boundary --manifold-surface --use-floodfill` [FFM_v20.stl__sf.zip](https://github.com/prusa3d/PrusaSlicer/files/12394008/FFM_v20.stl__sf.zip)

Thanks. Unfortunatly the slicer doesn't want to open the file. 

[image]

foreachthing, 2023-08-26:
Have you extracted the .obj from the zip file?
I can import it without message?!?

clawfire, 2023-08-28:
1. Unzip the file
2. Drag n Drop the .obj into a new slicer project
3. Got this error message. 

If I can collect some kind of logs from somewhere to help better understand what's the issue with this .obj file, I'll be more than happy to 😃

foreachthing, 2023-08-28:
> If I can collect some kind of logs

Sure, just start PS from command prompt: `prusa-slicer.exe -loglevel 5 > log.txt`
Then you got the log.txt with all kinds of messages. Maybe use loglevel 4 or even 3 ...

clawfire, 2023-08-28:
The log file is quite big

<details><summary>View the log file</summary>

``` txt
[2023-08-28 21:56:20.859162] [0x00000001e3711e00] [trace]   App updater default download path: "/Users/thibault/Downloads"
[2023-08-28 21:56:20.859373] [0x00000001e3711e00] [debug]   full path: 7484420837251052083
[2023-08-28 21:56:20.859381] [0x00000001e3711e00] [debug]   single instance: undefined. other params: ./PrusaSlicer;--loglevel;5
[2023-08-28 21:56:20.859398] [0x00000001e3711e00] [debug]   full lock path: /Users/thibault/Library/Application Support/PrusaSlicer/cache/7484420837251052083.lock
[2023-08-28 21:56:20.859479] [0x00000001e3711e00] [debug]   Creating lockfile.
[2023-08-28 21:56:20.859484] [0x00000001e3711e00] [info]    Instance check: Another instance not found or single-instance not set.
[2023-08-28 21:56:20.975985] [0x00000001e3711e00] [trace]   System language detected (user locales and such): fr_LU
[2023-08-28 21:56:20.988021] [0x00000001e3711e00] [trace]   Best translation language detected (may be different from user locales): fr
[2023-08-28 21:56:20.988113] [0x00000001e3711e00] [trace]   Switching wxLocales to fr
[2023-08-28 21:56:21.070351] [0x00000001e3711e00] [info]    Checking if indices need to be installed from resources...
[2023-08-28 21:56:21.483326] [0x00000001e3711e00] [info]    3dx mac handler starts
[2023-08-28 21:56:21.483417] [0x00000001e3711e00] [error]   3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 0x0005): tried: '/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/System/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file, not in dyld cache)
[2023-08-28 21:56:21.483425] [0x00000001e3711e00] [info]    3dx drivers loaded: NO
[2023-08-28 21:56:21.484453] [0x00000001e3711e00] [info]    Undo / Redo snapshot taken: Nouveau Projet, Undo / Redo stack memory: 0MB Resident memory: 192MB; Peak memory usage: 192MB
[2023-08-28 21:56:21.531457] [0x00000001e3711e00] [debug]   BuildVolume bed_shape clasified as: Rectangle
[2023-08-28 21:56:21.557185] [0x00000001e3711e00] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2023-08-28 21:56:21.557199] [0x00000001e3711e00] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2023-08-28 21:56:21.867854] [0x00000001e3711e00] [trace]   '3;90;0;0;43;<plist version="1.0"><dict><key>NSFontNameAttribute</key><string>Helvetica</string><key>NSFontSizeAttribute</key><real>13</real></dict></plist>'font descriptor string param of load_wxFont()
[2023-08-28 21:56:21.867882] [0x00000001e3711e00] [trace]   '3;90;0;0;43;<plist version="1.0"><dict><key>NSFontNameAttribute</key><string>Helvetica</string><key>NSFontSizeAttribute</key><real>13</real></dict></plist>' wx string descriptor
[2023-08-28 21:56:21.868101] [0x00000001e3711e00] [trace]   loaded font is 'Helvetica'.
[2023-08-28 21:56:21.868132] [0x00000001e3711e00] [trace]   input uri(file:///System/Library/Fonts/Helvetica.ttc) convert to path(/System/Library/Fonts/Helvetica.ttc) string(/System/Library/Fonts/Helvetica.ttc).
[2023-08-28 21:56:21.869953] [0x00000001e3711e00] [trace]   '3;90;0;0;43;<plist version="1.0"><dict><key>NSFontNameAttribute</key><string>Helvetica</string><key>NSFontSizeAttribute</key><real>13</real></dict></plist>' wx string get from GetNativeFontInfoDesc. wxFont IsOk(true), isNull(false)IsFixedWidth(false), IsUsingSizeInPixels(false), Encoding(43), 
[2023-08-28 21:56:21.900222] [0x00000001e3711e00] [debug]   TriangleMesh::repair() started
[2023-08-28 21:56:21.900690] [0x00000001e3711e00] [debug]   TriangleMesh::repair() finished
[2023-08-28 21:56:22.663099] [0x00000001e3711e00] [info]    Checking for cached configuration updates...
[2023-08-28 21:56:22.663134] [0x00000001e3711e00] [info]    Confing bundle not installed for vendor Rigid3D, skipping: 
[2023-08-28 21:56:22.663156] [0x00000001e3711e00] [info]    Confing bundle not installed for vendor TriLAB, skipping: 
[2023-08-28 21:56:22.663164] [0x00000001e3711e00] [info]    Confing bundle not installed for vendor Geeetech, skipping: 
[2023-08-28 21:56:22.663178] [0x00000001e3711e00] [info]    Confing bundle not installed for vendor Anker, skipping: 
[2023-08-28 21:56:22.663184] [0x00000001e3711e00] [info]    Confing bundle not installed for vendor Trimaker, skipping: 
[2023-08-28 21:56:22.663189] [0x00000001e3711e00] [info]    Confing bundle not installed for vendor Print4Taste, skipping: 
[2023-08-28 21:56:22.663194] [0x00000001e3711e00] [info]    Confing bundle not installed for vendor Elegoo, skipping: 
[2023-08-28 21:56:22.663199] [0x00000001e3711e00] [info]    Confing bundle not installed for vendor Snapmaker, skipping: 
[2023-08-28 21:56:22.665285] [0x00000001e3711e00] [debug]   Vendor: Templates, version installed: 1.0.3, version cached: 1.0.3
[2023-08-28 21:56:22.665300] [0x00000001e3711e00] [info]    Confing bundle not installed for vendor BIBO, skipping: 
[2023-08-28 21:56:22.665306] [0x00000001e3711e00] [info]    Confing bundle not installed for vendor LulzBot, skipping: 
[2023-08-28 21:56:22.682095] [0x00000001e3711e00] [debug]   Vendor: Prusa Research, version installed: 1.9.8, version cached: 1.9.8
[... cut ...]
par défaut	21:50:33.169853+0200	PrusaSlicer	order window: 5bc0 op: 0 relative: 0 related: 0
par défaut	21:50:33.205938+0200	PrusaSlicer	order window: 5ba9 op: 1 relative: 5bbf related: 0
par défaut	21:50:33.206007+0200	PrusaSlicer	order window: 5bbf op: 0 relative: 0 related: 0
par défaut	21:50:33.206604+0200	PrusaSlicer	order window: 5bbf op: 0 relative: 0 related: 0
par défaut	21:50:33.321021+0200	PrusaSlicer	order window: 5bc1 op: 0 relative: 0 related: 0
erreur	        21:50:33.321508+0200	PrusaSlicer	CGSWindowShmemCreateWithPort failed on port 0
```

closed 2026-09-29 by prusaslicer-bot[bot]