prusa3d/PrusaSlicer #11149: Repeated issues with "no suitable storage found" when trying to send gcode to mk4 over LAN

tverett, 2023-08-20:
### Description of the bug

70-80% of the time since using 2.6 prusaslicer "I get a no suitable storage found at <ip address>
" message. Sometimes it will work if I retry and most of the time it won't. Nothing has changed with regard to my LAN, USB stick used in the MK4 etc other than upgrading from previous beta version of 2.6 to released 2.6 and MK4 is now running RC input shaping firmware.  The MK4 recognizes the stick just fine and I can navigate the files on the stick without issue using the Mk4 menu/screen.

### Project file & How to reproduce

Try to send any file to MK4 using prusalink setup on LAN.
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 Arm64 MAC

### Operating system

Mac OS 12.4

### Printer model

Prusa MK4 running 5.0.0RC firmware

slicedbread, 2023-11-06:
I had this issue as well. I discovered this was because the printer had moved to a new IP address. My resolution was to just update the machine settings to target the new IP and I assigned the printer to have a static IP rather than a dynamic one.

If you're familiar with setting a static IP stop reading here, if you aren't here's a lil GPT generated tutorial:

Accessing Your Router
Find your router's IP address:

This is typically 192.168.1.1, 192.168.0.1, or something similar. You can often find it in the router's manual or on the bottom of the router itself. If you can't find it, you can often find your router's IP address by running ipconfig (on Windows) or ifconfig or ip a (on Linux/Mac) in the command line and looking for the "Default Gateway" address.
Log in to your router:

Open a web browser and enter the router's IP address into the address bar.
Enter the username and password to log in. This information may also be in the router's manual or on the router itself if you haven't changed it from the default. The default is often something like "admin" for both fields.
Setting a Static IP
Navigate to the DHCP settings:

Once logged in, look for a section called "DHCP", "LAN Setup", "Network Settings", or something similar.
Find the Address Reservation or Static Lease section:

Within the DHCP or LAN settings, look for an option to set a static IP, often called "Address Reservation", "DHCP Reservation", "Static Lease", or similar.
Choose your device:

You'll need to identify the device for which you want to set a static IP. This can usually be done by looking at a list of connected devices and their current IP addresses and MAC addresses. You might need to check your device’s settings to find out its current IP or MAC address to identify it on the router.
Enter the details:

Once you've identified the device, enter the IP address you want to assign to it. This should be within the local network's IP range (e.g., if your router's IP is 192.168.1.1, you might assign 192.168.1.100).
Enter the MAC address of the device, which is a unique identifier for the network interface.
Save the settings:

Once you've entered the IP and MAC addresses, save the settings. You might have to restart the router and the device for the changes to take effect.

tverett, 2023-11-06:
Yeah my printer is already set with a static ip address at my router so that’s not my issue. The issue has gotten better in the latest prusaslicer with beta mk4 firmware but I still occasionally get the issue. There seems to be an issue with the printer not always properly mounting/seeing the usb drive. I’ve noticed that if I reset the printer with the thumb drive inserted it won’t see it after it powers up until I remove it and reinsert it. When it’s in this state I get this error consistently and the only remedy is reinserting the usb drive.

ltpitt, 2024-01-14:
Same error here, using a prusa mini plus

tverett, 2024-01-27:
This error now occurs every time i try to send a file to my printer. I have a static IP address set already which hasn't changed since the day I installed my Prusa MK4. I've recently discovered that when this problem occurs if I open a terminal window and ping my mk4's IP address it will respond with no issues and then when I go back to prusaslicer i'm then able to send the file as expected. It's like Prusaslicer isn't properly checking the network for the presence of the Mk4 before it decides the printer is unreachable.

jandebleser, 2024-02-22:
I had this issue too. Turns out the USB drive was corrupt. Reformatting the USB stick as FAT32 fixed the issue.

JayKeyStk, 2024-02-25:
I've had this bug appearing every now and then, a quick restart have solved it. Up until yesterday. Could not upload anything to the printer, and could not ping my printer's IP address. No matter what I tried.

In my case, it was caused by setting the IP static!! gah! I bound the IP of both my Prusas yesterday, and I lost all connection to them. I've now unbound them both, and both are working again. 

So if anyone else has ZenWIFI XT9 from Asus, this could be your issue :-/

jbernalr, 2024-10-21:
This is happening to me as well. 
PrusaSlicer 2.8.1
MK3S+ (Firmware 3.14.0)
IP Addresses checked
Connection via PrusaLink (Browser, works)

jbernalr, 2024-10-22:
> This is happening to me as well. PrusaSlicer 2.8.1 MK3S+ (Firmware 3.14.0) IP Addresses checked Connection via PrusaLink (Browser) works

I tracked down the problem down to operating system (macos) permissions. 
Older versions of PrusaSlicer (tested 2.6.0) would cause a OS prompt to ask for permissions for PrusaSlicer to search the network. After you authorise this, you would have access to your printer. 
Newer PrusaSlicer Versions dont cause the prompt to come up, thereby you need another way to authorize searching the net.

dzioczek, 2024-12-14:
> I tracked down the problem down to operating system (macos) permissions. Older versions of PrusaSlicer (tested 2.6.0) would cause a OS prompt to ask for permissions for PrusaSlicer to search the [...]

I can confirm that this is the issue. I've went into Settings > Privacy & Security > Local Network, there I've allowed slicer to access network and everything is working as intended now. 

@jbernalr Thanks for this!

Razor412, 2025-02-28:
> > > This is happening to me as well. PrusaSlicer 2.8.1 MK3S+ (Firmware 3.14.0) IP Addresses checked Connection via PrusaLink (Browser) works
> > 
> > 
> > I tracked down the problem down to operating system (macos) permissions. Older versions of PrusaSlicer (tested 2.6.0) would cause a OS prompt to ask for permissions for PrusaSlicer to search the network. After you authorise this, you would have access to your printer. Newer PrusaSlicer Versions dont cause the prompt to come up, thereby you need another way to authorize searching the net.
> 
> I can confirm that this is the issue on macOS. I've went into Settings > Privacy & Security > Local Network, there I've allowed slicer to access network and everything is working as intended now.
> 
> [@jbernalr](https://github.com/jbernalr) Thanks for this!

This just started happening for me. I can access the printer through PrusaConnect but not PrusaLink. Only a reboot of the printer works for me. What version of MacOS are you on? I can't find that path in 14.7.2.

benvanik, 2025-03-09:
This is happening to me too now - I also cannot find "Local Network" settings. I've tried rebooting the printer and reformatting the drive to fat32, but each time I power up the printer I cannot print. Very bad :(

ndauten, 2025-04-03:
Hey Folks. I just had the same issue. A few things to test.

I think the core is the IP address does change and then things get misconfigured.

A few things to rule out classes of problems:

1. Can you `ping` the printer from the Terminal application on the same machine as PrusaSlicer: `ping ip_address` where the ip address is what you have in your local network. If you can see a response like `64 bytes from...` then your printer is on the network.

2. Make sure the settings for your physical printer in PrusaSlicer are correct. This took me awhile to find. 

  1. Go to Printers
  2. Click the dropdown box for your physical printer
  3. Click the little gear icon
[image]
  5. You should see a dialogue box that shows the network settings: Host Type, Hostname ..., Authorization Type, API Key / Password, ... 
  7. Make sure these are all correct
  8. Click test on the Hostname, IP or URL field button after enteringI found that my Wi-Fi router used DHCP and changed my printer address. After that things didn't work. I set the IP to statically assign and then went back to the settings. For some reason my API key had two characters that were lowercase when they should have been uppercase. I have no clue if: 1. things changed, 2. why it worked before, but I fixed this as well.

You do not need access to the local network for this to work *if you have the IP address of your printer*. I added a few screenshots to help you find the network configuration dialogue box.

Cheers!

FunkyBumbleton, 2025-07-10:
I have been trying to debug this issue for a while now. Here is what I've found.

- This issue occurred both when my printer was a MK4 and after I upgraded it to a MK4S.
- Issue happens over both WiFi and Ethernet.
-  Issue still occurs after assigning a static IP to the printer (both WiFi and Ethernet).
- Issue happens after trying with 3 separate USB drives.
- Two different but similar errors can occur in Prusa Slicer: Error 28 and Error 7
- Restarting the printer with the "Reset" button will temporarily fix the issue, but the issue returns about 10-20 seconds later.
- When the issue occurs, sometimes I can ping the printer from the terminal, but sometimes I can't. Even after successful pinging, the issue persists.
- When the issue occurs, I never can connect to the printer via PrusaLink (neither within the slicer nor from a separate browser).
- Sometimes, the issue fixes itself when I choose the "set ready" option on the printer. Although, when this works, I can only connect to it within the next 10-20 seconds.

I'm thinking this may be a firmware or hardware issue rather than an issue with PrusaSlicer, but I'm not an expert. Not sure what the odds are that this ever gets fixed, but this is a very annoying issue that has persisted for a long time.

### Environment Details
- **Slicer version:** 2.9.2
- **Firmware version:** 6.2.4  
- **OS:** Reproduced on Windows, Mac, and multiple Linux distros
- **Hardware:** Originally MK4, now MK4S (issue persists after upgrade)

der-eine, 2025-11-11:
Prusa Slicer 2.9.4 and MK4s 6.2.6 the problem still exists! In the meantime, 2 years have passed and still nothing has happened! I find this more than disappointing...

closed 2026-09-29 by prusaslicer-bot[bot]