prusa3d/PrusaSlicer #11150: Bridging don't anchor correctly

alanmakestuff, 2023-08-20:
I have been trying to print some bridges at the inner top of a dome shape. The bridges are extruded upside-down triangles so the peak (at the bottom) supports the above layers.. The issue is that the slicer prints the first string of the bridge before the perimeters so the anchors don't anchor the start and end of the bridge. The bridges fail at the first line.

[image]

Orcaslicer does it right in comparison, even putting little loops that add more anchorage. In orca you can even adjust the amount of the anchor loops to improve the bridge performance. I would like that in Prusaslicer please.

arshish1612, 2023-08-21:
Correct. I have observed the same issue.

ddacunha, 2023-09-12:
Same issues as #9999

Jan-Soustruznik, 2024-01-31:
Hello, @alanmakestuff , 

could you share your model or better project file (3MF), where problem can be reproduced, please? 

Thank you, have a nice day

alanmakestuff, 2024-02-01:
> Hello, @alanmakestuff ,
> 
> could you share your model or better project file (3MF), where problem can be reproduced, please?
> 
> Thank you, have a nice day

Hi @Jan-Soustruznik 

Here is the lamp base 3mf file in zip. However, this time when I sliced it in Prusaslicer it sliced it correctly... This version is 2.7.0. I cannot remember what the previous version was when I posted the issue.

[image]

[Base Outer.zip](https://github.com/prusa3d/PrusaSlicer/files/14121677/Base.Outer.zip)

lukasmatena, 2024-03-21:
Fixed in [PrusaSlicer 2.7.3-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-alpha1). Closing.