prusa3d/PrusaSlicer #11151: Bridging for small bridges broken

tuct, 2023-08-20:
### Description of the bug

I am printing the new Nevermore v6 and found a problem when slicing the model with PrucaSlicer 2.6 
The bridges do NOT work at all! i printed on a MK4 and a Voron 0.2.
When sliced with PS the bridges are generated as a small dot, the bridge and move to the next bridge:
[image]

Which results in this:

[image]

When printing the same file but sliced with Super slicer 2.5 the bridges are created in a continuous move which results in a much better bridging, see:
[image]

gcode and project files for Prusa and Super Slicer
[V6_Nevermore_bridging_issue_PS.zip](https://github.com/prusa3d/PrusaSlicer/files/12389240/V6_Nevermore_bridging_issue_PS.zip)

In Super Slicer 2.4 i had the same result as in PrusaSlicer 2.6

### Project file & How to reproduce

Check STLs and created gcode in zip,
print on MK4, SuperSlicer version works, PrusaSlicer not

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0+win64

### Operating system

Win11

### Printer model

MK4 and Voron 0.2

tuct, 2023-08-20:
I checked with PrusaSlicer 2.5.2 and got the same results as with 2.6

MrBlackmidi, 2023-08-22:
Duplicate of #9074

tastyratz, 2024-03-13:
I did a lot of digging on this issue and narrowed it down in the linked orcaslicer bug and reference the same bug downstream on superslicer as well.
This has to do with Arachne wall order when more than 1 perimeter is selected.

The solution would be "print detected bridges last/after perimeters" as a default behavior or checkbox for when bridges are detected. This should drastically improve bridging performance.

closed 2026-09-29 by prusaslicer-bot[bot]