prusa3d/PrusaSlicer #11152: Missing layer in tree supports

bh9140, 2023-08-21:
### Description of the bug

When trying to print on PRUSA mk4 - fw 5.0
profile IS 0.1 fast detail + tree supports
I had several failures. After investigating the matter, I noticed that after printing the first layer of the base, many support branches are not included in the slicer. They start printing only from 0.8mm and gradually new ones come up to 2.30mm.
Of course, this means that the supports do not stick to the ground and start to fly freely around the table. Print stop required.

I think it's an easy problem to fix. If the supports had forced contact with the bottom layer, everything would be fine.

[image]
[image]
[image]
[image]
[image]

### Project file & How to reproduce

Just file in attachment
[PRUSA bug report.zip](https://github.com/prusa3d/PrusaSlicer/files/12395668/PRUSA.bug.report.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 11

### Printer model

Prusa MK4

kubispe1, 2023-08-21:
It is duplicate. Thank you for your effort. It will take some time to improve organic support to resolve this issue. Thank you for your understanding.

kubispe1, 2023-08-21:
there is original issue: #9601

closed 2023-08-21 by kubispe1