prusa3d/PrusaSlicer #11157: After using 'Cut' tool in 2.6.0, unable to 'Add part' to object, thus can't add a 'Helper disk' to either of the two cut parts.

imacmill, 2023-08-22:
### Description of the bug

After using 'Cut' tool on a part in 2.6.0, unable to 'Add part' to either of the two cut sections. Thus, unable to add a 'Helper disk' to either of the two cut parts. The 'Add part' menu option is disabled on both cut parts.

### Project file & How to reproduce

Load a mode.
Cut it anywhere. Doesn't matter if you configure 'connectors' or not. Complete the cut.
Select and right-mouse click on either cut piece, 'Add part' is disabled. Needs to be enables to add, for example, a 'helper disk'.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0

### Operating system

System:   Kernel: 5.15.0-79-generic x86_64 bits: 64 compiler: gcc v: 11.3.0 Desktop: Cinnamon 5.8.4     tk: GTK 3.24.33 wm: muffin dm: LightDM Distro: Linux Mint 21.2 Victoria base: Ubuntu 22.04 jammy Machine:   Type: Desktop System: ASUS product: N/A v: N/A serial: <superuser required>   Mobo: ASUSTeK model: PRIME H570-PLUS v: Rev 1.xx serial: <superuser required>     UEFI: American Megatrends v: 0820 date: 04/27/2021 CPU:   Info: 8-core model: 11th Gen Intel Core i9-11900K bits: 64 type: MT MCP arch: Rocket Lake rev: 1     cache: L1: 640 KiB L2: 4 MiB L3: 16 MiB   Speed (MHz): avg: 1301 high: 3167 min/max: 800/5100:5300 cores: 1: 3167 2: 800 3: 1282 4: 809     5: 896 6: 1185 7: 1195 8: 1334 9: 872 10: 1010 11: 1756 12: 2100 13: 904 14: 973 15: 1426     16: 1114 bogomips: 112128   Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx Graphics:   Device-1: NVIDIA GA102 [GeForce RTX 3080 12GB] vendor: ZOTAC driver: nvidia v: 535.86.05 pcie:     speed: 16 GT/s lanes: 16 ports: active: none off: HDMI-A-1 empty: DP-1,DP-2,DP-3 bus-ID: 01:00.0     chip-ID: 10de:220a   Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: nvidia     unloaded: fbdev,modesetting,nouveau,vesa gpu: nvidia display-ID: :0 screens: 1   Screen-1: 0 s-res: 1920x1080 s-dpi: 92   Monitor-1: HDMI-0 res: 1920x1080 dpi: 92 diag: 612mm (24.1")   OpenGL: renderer: NVIDIA GeForce RTX 3080/PCIe/SSE2 v: 4.6.0 NVIDIA 535.86.05     direct render: Yes Audio:   Device-1: Intel Tiger Lake-H HD Audio vendor: ASUSTeK driver: snd_hda_intel v: kernel     bus-ID: 00:1f.3 chip-ID: 8086:43c8   Device-2: NVIDIA GA102 High Definition Audio vendor: ZOTAC driver: snd_hda_intel v: kernel     pcie: speed: 16 GT/s lanes: 16 bus-ID: 01:00.1 chip-ID: 10de:1aef   Sound Server-1: ALSA v: k5.15.0-79-generic running: yes   Sound Server-2: PulseAudio v: 15.99.1 running: yes   Sound Server-3: PipeWire v: 0.3.48 running: yes Network:   Device-1: Intel Ethernet I219-V vendor: ASUSTeK driver: e1000e v: kernel port: N/A     bus-ID: 00:1f.6 chip-ID: 8086:15fa   IF: eno1 state: up speed: 1000 Mbps duplex: full mac: <filter> Drives:   Local Storage: total: 6.41 TiB used: 41.02 GiB (0.6%)   ID-1: /dev/sda vendor: Western Digital model: WD Green 2.5 1000GB size: 931.51 GiB     speed: 6.0 Gb/s serial: <filter>   ID-2: /dev/sdb vendor: Western Digital model: WD4005FZBX-00K5WB0 size: 3.64 TiB     speed: 6.0 Gb/s serial: <filter>   ID-3: /dev/sdc vendor: Fanxiang model: S101 2TB size: 1.86 TiB speed: 6.0 Gb/s     serial: <filter> Partition:   ID-1: / size: 915.32 GiB used: 41.01 GiB (4.5%) fs: ext4 dev: /dev/sda3   ID-2: /boot/efi size: 512 MiB used: 6.1 MiB (1.2%) fs: vfat dev: /dev/sda2 Swap:   ID-1: swap-1 type: file size: 2 GiB used: 0 KiB (0.0%) priority: -2 file: /swapfile Sensors:   System Temperatures: cpu: 27.8 C mobo: N/A gpu: nvidia temp: 41 C   Fan Speeds (RPM): N/A gpu: nvidia fan: 47% Repos:   Packages: 2452 apt: 2424 flatpak: 28   No active apt repos in: /etc/apt/sources.list   Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list     1: deb https: //mirrors.ocf.berkeley.edu/linuxmint-packages victoria main upstream import backport     2: deb https: //mirror.it.ubc.ca/ubuntu jammy main restricted universe multiverse     3: deb https: //mirror.it.ubc.ca/ubuntu jammy-updates main restricted universe multiverse     4: deb https: //mirror.it.ubc.ca/ubuntu jammy-backports main restricted universe multiverse     5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse Info:   Processes: 384 Uptime: 2d 15h 50m Memory: 31.17 GiB used: 9.99 GiB (32.1%) Init: systemd v: 249   runlevel: 5 Compilers: gcc: 11.4.0 alt: 11/12 Client: Cinnamon v: 5.8.4 inxi: 3.3.13

### Printer model

Creality Ender 3 V2

YuSanka, 2023-08-30:
Thanks for your report. But that's actually expected behavior.  
When you cut without connectors, then you get two separate objects. And you can manipulate with them whatever you want.
But if connectors was added, than result objects are related. And to avoid brake this relation we suppress to add parts/negative volumes and a separate scale of the related parts.

closed 2023-08-30 by YuSanka

mark-a-scott, 2023-12-28:
PrusaSlicer Team, would you please reconsider this behavior? The way the current behavior is, makes it so that large, flat models, that tend to warp cannot be printed. I have designed a base plate for a model that I've had to cut into sections to minimize warping. Because I've used connectors to re-align it during assembly, I cannot use "mouse ears" to help with keeping the corners glued to the print bed. 

My model is 220mm x 130mm x 10 mm, with 25% infill. I want it to have some weight to it. I am currently unable to print it even though my MK4 is in an enclosure.

RubenVerster, 2024-10-16:
@YuSanka I also agree with @mark-a-scott 

I'm in a similar situation where I cannot add a helper disc to my dovetail cut part ':)
It would have helped so much :')

I think I'll just export the cut part for now and add mouse ears manually