prusa3d/PrusaSlicer #11159: Reset Snap Connectors [CUT]

wwian, 2023-08-22:
**Is your feature request related to a problem? Please describe.**
Snap Connectors [CUT] 

**Describe the solution you'd like**
Reset snap connectors settings to standard/ original.

**Describe how it would work**
A button or other function to reset snap connector setting to standard/ origianl.

**Describe alternatives you've considered**

**Additional context**

[image]

YuSanka, 2023-08-30:
Thanks for your report. (SPE-1854)

lukasmatena, 2023-09-05:
Implemented in [2.6.1-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.1-rc2). Closing.

closed 2023-09-05 by lukasmatena