prusa3d/PrusaSlicer #11160: After color change, printer leaves a dot on random position before resuming print

majedation, 2023-08-23:
### Description of the bug

I've added a color change in the slicer through the layer bar on the right side.
When the printer reaches the filament change layer and I change the filament, the printer then resumes by printing a dot on a random spot before resuming with the new layer.
This happened with me couple times with different parts.

[image]
[image]

### Project file & How to reproduce

[Color Change Bug.zip](https://github.com/prusa3d/PrusaSlicer/files/12414988/Color.Change.Bug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

macOS 13.4.1

### Printer model

Prusa i3 MK3S

r3Fuze, 2023-08-23:
Related PR: https://github.com/prusa3d/PrusaSlicer/pull/9036

rtyr, 2024-02-18:
Fixed in 2.7.2-alpha2.

closed 2024-02-18 by rtyr