prusa3d/PrusaSlicer #11162: Arrangement ignored the following objects which can't fit into a single bed

war4peace, 2023-08-23:
### Description of the bug

I have a model which is 155x142x100mm, placed on a 180x180x180mm printer bed. The object fits well inside the printer volume.
When I click the Arrange button, I receive the message in the title:

`Arrangement ignored the following objects which can't fit into a single bed`

[image]

Printer is a RatRig V-MiniON.

### Project file & How to reproduce

Zip archive is too large and the model is proprietary, I could share it separately if needed.
However, any object with the same dimensions throws the same error, even a cubic shape.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0+win64

### Operating system

Windows 10 fully updated

### Printer model

RatRig V-MiniON

5axes, 2023-08-23:
Have you activated the option : **_Enable Rotations_** in the Arrange options ?

[image]

war4peace, 2023-08-23:
The same error with or without the option enabled, I apologize, I should have mentioned that.
It shouldn't have mattered, though, because in my screenshot the model has more than enough space around it, and it was almost centered, I wanted to place it directly in the center.

I have attached a simple project, just an added shape (box)  with the same dimensions, to replicate the bug. The error is thrown with that shape only, placed exactly in the middle of the bed.

[box_ps_bug.zip](https://github.com/prusa3d/PrusaSlicer/files/12421789/box_ps_bug.zip)

[image]

jmd73, 2023-08-23:
@war4peace It's not actually a bug. It takes the skirt and brim into account.
If you disable skirt or brim, it arranges fine.

war4peace, 2023-08-23:
For the cube, indeed, that's the case.
For the initial model I screenshotted, not anymore.
When slicing, the brim and skirt are all within the bed size. there is no toolpath outside the volume. As you can see in the screenshot, the object is not centered, but can't be centered because PS still complains.

If I disable both skirt and brim, it would center.

I guess it's a combination of factors. At any rate, maybe the error message should also contain the following text: "Try disabling skirt/brim".

[image]

tamasmeszaros, 2023-08-30:
Hi @war4peace ! 
Does this problem still persist in version 2.6.1-rc? if it does, please check your "spacing from bed" setting (see screenshot from @5axes with the arrange dialog) as that could render the model unarrangeable if set too high.

closed 2026-09-29 by prusaslicer-bot[bot]