prusa3d/PrusaSlicer #11164: Organic Supports generated outside the print area

OttScott, 2023-08-23:
### Description of the bug

Organic supports are generating outside the print are for large prints. Space exists in other places, but the biggest issue is that toolpaths should never exceed the print volume.
[image]
[image]

### Project file & How to reproduce

Common community print 'GEN2 Steel Sheet Holder V2 for Lack Enclosures' on Pritables.com 
((https://www.printables.com/model/295913-gen2-steel-sheet-holder-v2-for-lack-enclosures/files))
[BUG_REPORT - Back Side - GEN2 Steel Sheet Holder V2 - Prusa.zip](https://github.com/prusa3d/PrusaSlicer/files/12420893/BUG_REPORT.-.Back.Side.-.GEN2.Steel.Sheet.Holder.V2.-.Prusa.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 11 (22H2 - Build 22621.2134)

### Printer model

Prusa Mk4

hatesyourface, 2023-10-12:
Similar issue on the XL; with a model the full width of the bed the right side had the organic support in front of the model, but on the left side (it's a symmetrical object) the support was generated outside the printable area and I was unable to orient the part in a way that it would be usable. The supports weren't symmetrical, even though the object to print was. Either they need to be generated inside the print area or we need the ability to decide where the "trunk" will originate around the object.

ghylander, 2023-11-13:
Experiencing this as well, but not limited to the trunk base:

[image]

closed 2024-04-12 by Jan-Soustruznik