prusa3d/PrusaSlicer #11170: Arbitrary build volume shape

Tupson444, 2023-08-24:
2.6.1-rc1 has introduced arbitrary (custom) bed shape. Similar to that, would it be possible to introduce defining arbitrary build volume shape?

Like importing 2D STL file for defining custom bed shape, there could be an option to import 3D shape of the build volume.

This can be useful if using bed clips, which should be avoided up to, say 5mm, while overhangs leaning above the clips should be permitted. Another case is, for example, if a printer has certain Z max height, but one corner should be avoided to prevent collision with print frame (this applies to custom modified printers obviously, as factory ones shouldn't have such a problem).

kubispe1, 2023-08-28:
You are right. In general, there are some applications. Delta printers, I have in my mind.

Tupson444, 2026-09-14:
still relevant

Jan-Soustruznik, 2026-10-20:
Hello, thank you for the update. Following our new [policy,](https://github.com/prusa3d/PrusaSlicer?tab=contributing-ov-file#issue-tracker-policy) I am closing this feature request and converting it into a Discussion thread. This will allow us to hear what the community thinks about this feature and provide a place for others to upvote it. Have a great day.