prusa3d/PrusaSlicer #11171: Arbitrary bed shape limitation (2.6.1-rc1)

Tupson444, 2023-08-24:
I'm sure that developers are aware of this, and some improvements might be already planned, but I'm opening this issue to discuss possible improvements for custom bed shape (introduced in 2.6.1-rc1).

Whatever the bed shape is, arranging works properly. Objects are placed only inside the boundaries of the custom bed shape, even if it is concave:
[image]

However, when manually moving an object, it is possible to place it outside of bed boundaries:
[image]

That is, the boundary for manual placement is convex, because objects can't be placed outside of that:
[image]

Also, travel moves are not restricted by concave shape:
[image]

Because of this, custom bed shape can't reliably be used to avoid accidentally placing parts in non printable area, or, for example, avoid hitting bed clips.

Otherwise, this is a great new feature. For example, it can already be used to avoid auto arrange over damaged areas of bed surface (I haven't tried that yet though).

kubispe1, 2023-08-28:
Thanks for the feedback, I forwarded it to the developer.

Tupson444, 2026-09-13:
Seems to be resolved in 3.0.0 alpha (except that travel moves are not restricted by bed shape, but that can be considered a different issue).

closed 2026-09-13 by Tupson444