prusa3d/PrusaSlicer #11172: new feature

sz3bbyla, 2023-08-24:
### Description of the bug

I ask my self if is posiible in slicer for end of slice we have an list with all kind of type of slicing, the theory is if can be managed in time printing, like to change solid infil procent or top solid infil , or gap infil, like lableling this types of slicing and cahange some labels in function of needs in time printing

### Project file & How to reproduce

there no error , just an question

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.59.2

### Operating system

ubuntu 22

### Printer model

voron.2.4 350 cw2

jmd73, 2023-08-25:
@sz3bbyla  If I remember correctly, this has been suggested before.
Maybe this can be accomplished with slicing via command promt, with a script. (I don't know enough about scripting, to suggest how to do it.)

And the version you have given in seems to be a SuperSlicer version, so maybe you meant to request the function there?

jmd73, 2023-08-25:
@sz3bbyla Found one that is similar, but not completely: #6637

hlovdal, 2023-08-25:
@sz3bbyla Can you update the issue to have a more descriptive title than "new feature"?

kubispe1, 2023-08-28:
Thanks for the feature request and your idea. I will let open #6637 and close this one with the link. There is a more detailed description.

closed 2023-08-28 by kubispe1