prusa3d/PrusaSlicer #11173: Top layer bridge is in air

pasadei, 2023-08-24:
### Description of the bug

It has no overlap and is printed in air when the bridge is the top layer.
[image]

### Project file & How to reproduce

[Bridge.zip](https://github.com/prusa3d/PrusaSlicer/files/12431992/Bridge.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 10, 22H2

### Printer model

Ender 3 s1

kubispe1, 2023-08-30:
PS slicer at the moment (2.6.1-rc1)  does not expand bridges to top solid infill.

ddacunha, 2023-09-12:
Looks related to [#9999](https://github.com/prusa3d/PrusaSlicer/issues/9999)

lukasmatena, 2024-03-21:
Fixed in [PrusaSlicer 2.7.3-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-alpha1). Closing.

closed 2024-03-21 by lukasmatena