prusa3d/PrusaSlicer #11174: When slicing on the new 2.6.1 I get error "PS has encountered a fatal error: "Access Violation" Please save and restart

joseuzair718, 2023-08-24:
### Description of the bug

[image]

I get this error when trying to slice this shape. I used shapes from the slicer itself to test out the new text feature so I thought I'd make a pencil holder with my company name and number on it. (Also as a side note, it actually allows me to use a wipe tower now rather than a draft shield for multi material, before, ALL multi material projects were forcing me to use a draft shield so I'm glad to see that fixed and can now use a wiper tower)
But anyway, when I slice, I get this error. Not sure what's causing it. I'm using my work laptop so when I reach home I'll use my own laptop and see if the problem persists. However when I searched for this issue I only found it from about a year ago and no mention of it since then so I'm thinking this update may have something to do with it?
Has anyone else experienced this error with this new version? I've attached the 3MF file in a ZIP if anyone wants to give it a try, how all I've done it create two cylinders, one is a negative to create the inside of the pencil holder, then added text on the outside...

I didn't get this error before I started setting up the profiles for my printer (since I'm on my work laptop) so many one of those settings is causing this, which is why I'll have a better idea as to what's causing this when I reach home and try it on my own laptop but if anyone can point me in the right direction I would be much appreciated.

Thanks!

Note: I know I set up as the Artillery Sidewinder X1 just to make it easy as most settings are the same, other than the fact that the SV-06 is an IDEX but I just wanted to test this new TEXT feature in PRUSA and then slice it to see how it would come out... However if many people think this could be the issue right off the bat, I'll try it on my laptop which has all profiles set up properly for multi colored prints and what not... (Although I have used gcodes for single extruder prints for my Sidewinder X1 directly on my SV-06 and it printed just fine. But let's see what happens!

[Cylinder Error.zip](https://github.com/prusa3d/PrusaSlicer/files/12432141/Cylinder.Error.zip)

### Project file & How to reproduce

[Cylinder Error.zip](https://github.com/prusa3d/PrusaSlicer/files/12432209/Cylinder.Error.zip)

Open 3MF file, slice, and the error pops up right away, thus not allowing me to export the gcode to go through the gcode to review individual layers...

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1-rc1+win64

### Operating system

Windows 11

### Printer model

Sovol SV-06 (Also have Sidewinder X1 and X2)

joseuzair718, 2023-08-24:
[image]

Ok so if I take off the wipe tower, I instead get this error... Deleting these two files while Prusa Slicer is closed doesn't do anything, the error will come back...

However if I remove the wipe tower AND the draft shield, then it'll slice just fine... However that's obviously not ideal for a multi material print.

At this point I think it's the profiles I'm using so when I get home and try it on my own laptop I'll give an update... Hopefully a positive one...

joseuzair718, 2023-08-24:
Actually if I try to use a draft shield, I'll get the error, but then I save and close Prusa Slicer, delete those files, open up Prusa and open the 3mf and it slices just fine...

So it seems like Prusa Slicer wants me to only use a draft shield for multi material projects which is what the older version forced me to do, I made a post on reddit about it. It never gave me errors, the older versions just ignored the fact that I had draft shield disabled and wipe tower enabled, but when I sliced it would still make a draft shield... Which I just learned to deal with when printing multi material or multi colored prints...

Maybe this version is trying to change that but it's giving errors when trying to do so...

Again, I'll try from my laptop at home, I just thought I'd give updates as I learn them... It could be the profiles I'm using so maybe my laptop at home will do everything properly...

kubispe1, 2023-08-29:
We are looking into it. Thanks 
SPE-1849

FidelCapo, 2023-09-04:
This issue should be fixed and the fix will be part of the next release.
Closing.

closed 2023-09-04 by FidelCapo

joseuzair718, 2023-09-05:
Awesome, Just wanted to say thanks!!!! Prusa seems to be on top of the game right now!!!
Keep it up!!!!

Got me to switch from Cura to Prusa Slicer for my Sovol 6 IDEX and now using it for ALL my printers!!!!

protostheticspope, 2025-05-07:
This issue is back/never got fixed/got un-fixed in 2.9.0 - Getting the error when slicing multimaterial prints. :/