prusa3d/PrusaSlicer #11178: Extra Length on restart after retract not working

how-eee, 2023-08-25:
### Description of the bug

Extra Length on restart after retract not applied to gcode when "wipe while retracting" checkbox is on.

[image]
[image]

### Project file & How to reproduce

open any stl
add retract amount and Extra Length (negative in my case)
check gcode
switch on wipe
check gcode

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1 rc1

### Operating system

win 10  64

### Printer model

my own corexy

pedroloco2, 2023-08-25:
Extra length on restart allows only positive values, any value below 0 is equal to 0.

how-eee, 2023-08-26:
> Extra length on restart allows only positive values, any value below 0 is equal to 0.

That is wrong. Negative values works fine when wipe is off.

Why is negative values prohibited? Something wrong with mathematics there?

pedroloco2, 2023-08-26:
> > Extra length on restart allows only positive values, any value below 0 is equal to 0.
> 
> That is wrong. Negative values works fine when wipe is off.
> 
> Why is negative values prohibited? Something wrong with mathematics there?

With Wipe while retracting active (as you asked), Extra length on restart allows only positive values, any value below 0 is equal to 0.
In fact, it would not be too logical that there being no or very little pressure in the extruder (which is the function of Wipe while retracting), it is intended to add an additional "retraction". This can reduce the seam, but at the cost of leaving a "hole".
In any case, it would be interesting to add an option that allows determining the amount in mm of Wipe while retracting, which would allow greater control in cases where it is needed.

InstantBenHur, 2023-09-03:
Positive values don't work either, see #9902 . Just checked version 2.6.1-rc2, problem is still there.

patrick505, 2023-09-12:
I also have this issue. Extral Length doesn't work when wipe while retracting is activated. Please fix this bug ASAP. I have hundreds of single-wall models that I need theses settings for. Thanks

experiment322, 2024-07-10:
I'm also encountering this and, as a workaround for now, you can set `Retract amount before wipe` to something bigger than 0%, even 1%, and extra length on restart will start working again.

Wegerich, 2025-04-21:
I'm still having this problem in v 2.9.2. It seems like something about "wipe" is broken. By turning it off I was able to have good performance with retractions to prevent stringing, without needing wipes.

closed 2026-09-29 by prusaslicer-bot[bot]