prusa3d/PrusaSlicer #11180: Supress Warnings in GUI

sgano, 2023-08-25:
**Is your feature request related to a problem? Please describe.**
When slicing parts I often get annoying overlay warning messages like "WARNING: Detected print stability issues:[More]".  While I can see how that would be helpful to a beginner; I find it really annoying every time I reopen files or make a new project etc.   Please consider a way to disable the different types of warnings like this.  

**Describe the solution you'd like**
suppress warnings options

**Describe how it would work**
see above

**Describe alternatives you've considered**
-

**Additional context**
-

sgano, 2023-08-25:
Example of warnings I don't need to see when opening a file: 
[image]

(1) I don't need to see there is a new RC version released (I only am interested in actual releases) 
(2) I don't need to have the stability warning, everytime I open the file.  In fact I don't really need it at all as I normally check for this issues visually - and often this warning is wrong.  
(3) I don't need the objects were loaded with variable layer height -- this is obvious from the items in the right panel (under expert mode they all show "Variable layer height" anyway.  Why clutter the UI with these every time I open the file?

JohnOCFII, 2023-08-26:
> (1) I don't need to see there is a new RC version released (I only am interested in actual releases)

This one (at least) is configurable in the Preferences page.
[image]

sgano, 2023-08-28:
Excellent, I missed that option about new release notifications!   This same option for the other warnings and informational items (like the variable layer heights) would improve the user experience greatly.

neophyl, 2023-08-28:
You can also disable the support warnings already, in the preferences section.  Preferences>General>Alert When supports needed.

sgano, 2023-08-28:
Thanks! That may help with the stability warnings (I need to test that more); though by the name of that option it wasn't clear that it also applied to stability warnings.  The "variable layer height" info messages (which always require the user to hit the "x" to make them go away) don't seem to have an option to be disabled.

kubispe1, 2023-08-29:
@sgano this issue seems to me that is answered. I am closing. In case, let me know.

closed 2023-08-29 by kubispe1

sgano, 2023-08-29:
@kubispe1 - while parts are closed at least one major portion remains open: The "variable layer height" info message that shows up when opening a project that has been sliced with variable layers.   There seems to be no current way to turn that warning off.