prusa3d/PrusaSlicer #11183: [REQUEST] Add % shrinkage compensation to filaments tab

MarkoAnte, 2023-08-25:
Hi!

**Is your feature request related to a problem? Please describe.**
Problem I'm facing daily. I run a print farm. I have prusas and vorons. Lets say a client sends a box of 200x200 mm if I print this box as is, after cooling it turns out 199x199 mm if printed in ABS and about 199.4 if printed in PLA. For smaller items this is not a problem, but for e.g. larger electronics box, now screws don't fit into pcb etc.  When getting into larger print parts like the vorons or prusa XL support, the error due to material temperature shrinkage can be in multiple mm range. 

**Describe the solution you'd like**
I would like to have a per filament x/y shrinkage compensation, that scales the x/y dimensions automatically. 

**Describe how it would work**
After doing a test print, you can do a actual_dimension / wanted_dimension and get some number. Tho the factor changes with infill, speed, temp it will be in the sameish ball part for the same material. E.g. ABS at 99.4%, PLA 99.6% PA12 99% 

This is not x/y compensation where it adds extra wall thickness, just pure % scaling. 

**Describe alternatives you've considered**
When I was using PrusaSlicer I used to select the part, deselect scale uniformly and scale the part to e.g. 100.6%. I would often forget to to this. 
I have switched to SS for primarily this reason, but its development seamed to have stopped. 

Please prusa team, make getting more accurate dimensions easier.

5jvm0u4, 2023-10-19:
Same request here, when printing ABS and PC, I always find it shrinks noticeably. I'm now adding compensation manually, but it is sometimes annoying, and the value needed differs between each filament, so I also think that adding this to the filament tab will be very useful.
I also wish that a general shrinking compensation value(in mm) could be set in printer setting because I observe a constant shrink value of 0.2mm of all my print with any size at any location.

Phil1988, 2024-03-17:
SuperSlicer introduced this like 4 years ago as "filament_shrink" parameter in the Filament section:  (https://github.com/supermerill/SuperSlicer/issues/287).

I honestly dont understand, why Prusa avoids it so hard to implement this, because it is very handy and I use it on a daily basis.
It works very well in X-Y axis and also if its not perfect (because it doesnt take the z-axis into account) the alternative not having anything is just a huge drawback against all the other slicers.
You **need** this at least for ABS, ASA and Nylon (PA6/12)

SuperSlicer has it.
Bambu Studio took it from SuperSlicer, so its having this.
Orca Slicer also took it from SuperSlicer, so its having this.

Also +1 for this.

Xing777, 2024-04-24:
I would like this feature for certain filaments as well (ABS, Nylon, PC etc.), it needs to be in the filament settings.

HonzaJaros, 2024-04-30:
Is this happenin??

MarkoAnte, 2024-04-30:
> Is this happenin??

You see how old this thread is right  :) 

I don't think so. I think prusas mentality is "Our printers print perfect and are on size" I saw a FB post in their group of somebody asking how to shrinkage compensation and he was chewed up that prusas print ON SIZE. 

Super Slicer is back to being updated and worked on. Check it out.

Phil1988, 2024-04-30:
SuperSlicer is still far behind because supermerill was gone for about a year.
the development is now not as free as before, as he develops things for a company first and 1.6k issues open.

Prusas view in this regard is not understandable.
I loved the company, but not only the printers lost track to the standard but their slicer also is missing important settings they could easily include because ist already done.

looks like they got too big and dont care anymore for the community.

HonzaJaros, 2024-04-30:
I wonder if someone from community can do it and let them review and implement, isn’t it GitHub open source just for that?

Phil1988, 2024-04-30:
Its more like a matter of minutes.
The basis is identical so its really copy and paste for that code snippets.

@HonzaJaros This would only make sense if someone forks it and maintain it.
No one is doing a pull request, because they would deny it.
If they wanted this feature, they included it years ago in a couple of minutes.

gsasha, 2024-06-18:
This is the primary feature that makes me use Orca Slicer, while I'd very much prefer Prusa Slicer.

> I wonder if someone from community can do it and let them review and implement, isn’t it GitHub open source just for that?

There is already a pull request that implements it: https://github.com/prusa3d/PrusaSlicer/pull/11549

HonzaJaros, 2024-06-18:
It’s in the new alpha

Rudd-O, 2024-08-10:
This already shipped.  I used it today.

Rudd-O, 2024-08-10:
Can this ticket be closed?  I can see shrinkage compensation for the Z and the X/Y axes in the Filaments tab.

gsasha, 2024-08-11:
Yes, it works.

VH-fixer, 2025-04-07:
The professional fix is to use the thermal expansion value of the filament and have it filled in automatically.  Having a user add a shrinkage factor is no different than using the scale value.  Same process, test and build your own database.    This should be part of the filament definition.   It should be an on/off switch for the user to enabel and run from the filament selected.    A user over-ride should be an option, not the only way to use it.

SachCZ, 2025-04-08:
Hi @VH-fixer if you want you can create a feature request and describe in more detail what you suggest. The OP issue seems to be resolved, closing.

closed 2025-04-08 by SachCZ