prusa3d/PrusaSlicer #11186: Segmentation fault on object deletion

craigcabrey, 2023-08-26:
### Description of the bug

When removing an object from the plate, PrusaSlicer reliably segfaults.

### Project file & How to reproduce

1. Open project file
2. Right click model > delete (or use delete keyboard shortcut)
3. Crash with following log message (last one before getting a shell back):
```
[2023-08-26 12:09:45.535571] [0x00007f0df45db640] [info]    Exporting G-code finished Resident memory: 520MB; Shared memory: 125MB; Private memory: 395MB; Peak memory usage: 520MB
[2023-08-26 12:09:48.198577] [0x00007f0e80af7d40] [info]    Undo / Redo snapshot taken: Delete Selected Objects, Undo / Redo stack memory: 0MB Resident memory: 520MB; Shared memory: 125MB; Private memory: 395MB; Peak memory usage: 520MB
/usr/include/c++/11.3.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = Slic3r::ModelObject*; _Alloc = std::allocator<Slic3r::ModelObject*>; std::vector<_Tp, _Alloc>::reference = Slic3r::ModelObject*&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
```

[Example Crash.zip](https://github.com/prusa3d/PrusaSlicer/files/12446145/Example.Crash.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 Flatpak

### Operating system

Fedora Silverblue 38

### Printer model

Ender 3 S1 (Klipper)

PCGamesF1Tech, 2023-10-19:
I can confirm this bug under Linux Mint 21.2 Cinnamon and LMDE6 recently, the slicer window just closes.
Adding any other object before deleting (DEL key here) the unwanted one prevents it from doing so.

chmurli, 2023-10-27:
Hello,
I had same crash (so annoying) for many days.
Finally I have found that this crash is because 'auto-center parts ' feature!

**So just disable "auto-center parts" feature in preferences.**
I guess application wants to center object(s) but the list is empty ;D.
Hope it will be fixed soon the proper way ;).

KeDL01, 2023-11-13:
Same issue on linux mint 21.2 Cinnamon with version 2.6.1. Can confirm disabling "auto center parts" fixes the bug.

lukasmatena, 2023-11-16:
Fixed in [2.7.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-rc1). Closing.

closed 2023-11-16 by lukasmatena