prusa3d/PrusaSlicer #11188: Crash on replace with .stl after drawing seam when replaced by lower poly count

gaiuscosades, 2023-08-26:
### Description of the bug

On my machine prusaslicer crashes consistently after drawing on a seam and then replacing the object by a different .stl with a lower polygon count.

### Project file & How to reproduce

[replace_crash_template.zip](https://github.com/prusa3d/PrusaSlicer/files/12446499/replace_crash_template.zip)

1. open the .3mf provided
2. enter draw seam mode on the sphere object and draw additional seam positions (will not happen with saved project if not drawn after loading)
3. see that the object can be sliced successfully
4. replace the object by the .stl provided (cube)
5. move cube to center (is there a good reason tha one cannot draw seam outside of the printable area?)
6. enter the draw seam mode again for crashing the application

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0+win64-202306191221

### Operating system

Windows 11 Home Build 22621

### Printer model

MK3

lukasmatena, 2023-08-28:
Thanks, looking into it. It is not a new bug, it is already there in 2.4.2.

gaiuscosades, 2023-08-28:
Sorry, if I opned a duplciate case. I could not find a similar issue, maybe because of the wrong keywords used.
Cheers.

lukasmatena, 2023-08-28:
@gaiuscosades I did not mean that it was already reported, it probably (quite surprisingly) wasn't. What I meant is that it was already broken before the new 2.6.1-rc1 release. Thanks for the report.

gaiuscosades, 2023-08-28:
I understand, you just bisected back to older versions to see when it was intoduced, thanks for the clarification!
Happy to help you with quality assurance, for this great tool!

FidelCapo, 2023-08-29:
This issue should be fixed and the fix will be part of the next release.
Closing.

closed 2023-08-29 by FidelCapo