prusa3d/PrusaSlicer #11189: Auto Arrange doesn't adjust for height

georgewoodall82, 2023-08-26:
### Description of the bug

In the new version, auto arrange now can move models inside each other which is great but it doesn't put models under each other.

Screenshot:
[image]

Arranging models under each other would allow for even more models to be fit on one build plate.

### Project file & How to reproduce

[project.zip](https://github.com/prusa3d/PrusaSlicer/files/12446898/project.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.1-rc1+linux-x64-GTK3

### Operating system

arch btw

### Printer model

Voxelab Aquila

tamasmeszaros, 2023-08-28:
Thank you for the suggestion. Nice feature for some distant future version of PS.

antimix, 2023-09-01:
But what happens if you generate the support ? It will need that space under.

georgewoodall82, 2023-09-01:
it should only adjust objects underneath angles that are above the overhang threshold.

closed 2026-09-29 by prusaslicer-bot[bot]