prusa3d/PrusaSlicer #11190: Why are seams of version 2.4.0 much much better?

ramosglauco, 2023-08-27:
### Description of the bug

Hello. Since version 2.4.0 the seam pattern has been changed to look for smoother lines. However, this pattern fails on two main points in my opinion:
1 - When trying to follow an aligned path on models with curved surfaces (IMAGE1).
2 - When trying to hide the seams in the sharpest corner of models with more complex geometries (IMAGE2).

I work with the manufacture and sale of decorative objects, where the aesthetic part is very very important. For this reason I'm still stuck on version 2.4.0 and unfortunately I end up not being able to enjoy the new features of the newest versions. I would like to know if there is any idea of ​​reworking the seam algorithm. Thanks.

**IMAGE1 - PRUSASLICER 2.4.0 x 2.6.0**
[image]

**IMAGE2 - PRUSASLICER 2.4.0 x 2.6.0**
[image]

### Project file & How to reproduce

[seamexemple.zip](https://github.com/prusa3d/PrusaSlicer/files/12447045/seamexemple.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.6.0

### Operating system

Win11

### Printer model

FLSun SR

stefan-am, 2023-08-31:
I also agree that the seam placement in versions later than 2.4.0 often does not work according to what the user wants to achieve :-( 

Below is a simplified example of a handle, where i want to place the seam - for the reason of symmetry - in the middle of the rear face.  
This was  easy in Slicer 2.4.0, set seam to "Rear" and the seam came out in a exact vertical line in the middle of the rear face: 
 
[image]

The behaviour changed to the result shown here with 2.6.1RC1:

[image]
I want to be able to achieve the former behaviour of version 2.4.0 for the reason of a symmetrical look. 
Here the project file: 
[seam_rear_version_240.zip](https://github.com/prusa3d/PrusaSlicer/files/12491471/seam_rear_version_240.zip)

ramosglauco, 2023-09-02:
There are so many new features, but they forget to fix the basics.

nvtkaszpir, 2023-09-23:
Also looks like some seams are getting shifted, so seam is no longer a straight line byt is is moved with specific layers in one direction - rounding errors?

https://github.com/prusa3d/PrusaSlicer/assets/1480252/00908cb1-5bea-49a1-8bea-dd5746101ce1

ramosglauco, 2023-09-24:
They've known the seams are shit for a long long time. But they are worried about other useless things.

tabletopstor3d, 2024-03-23:
+1 would love to see this fixed. We recently jumped from 2.4 to 2.7 in order to support our XL and its been a huge step backwards.

Jan-Soustruznik, 2024-09-16:
closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/8443