prusa3d/PrusaSlicer #11192: Output options > Complete individual objects in PrusaSlicer 2.6.1-rc1 is not working

OTTO3D, 2023-08-27:
### Description of the bug

Option described below when is checked is not working properly when click in Arrange [A] in Plater section.
_Output options > Complete individual objects_ in PrusaSlicer 2.6.1-rc1 is not working
The objects are not being distributed automatically.
[image]

### Project file & How to reproduce

[Rota da Luz 200x200.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/12447169/Rota.da.Luz.200x200.stl.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1 rc1

### Operating system

macOS Catalina 10.15.17

### Printer model

Creality CR6-Max

kubispe1, 2023-08-28:
Is it a duplicate issue? Check #11167

OTTO3D, 2023-08-28:
Sorry not to check issues before submitting a new one.

kubispe1, 2023-08-29:
don't worry. I am closing this issue as duplicate

closed 2023-08-29 by kubispe1