prusa3d/PrusaSlicer #11194: Ooze prevention sends a ridiculous temperature. M104 S4294967276 T1

stonemandave, 2023-08-27:
### Description of the bug

When using ooze prevention to cooldown inactive hotend sends dangerous temperature numbers. M104 S4294967276 T1
I expected temperature - 30.

### Project file & How to reproduce

[dual_color_cali-cube_-_v2_-_s1.zip](https://github.com/prusa3d/PrusaSlicer/files/12448228/dual_color_cali-cube_-_v2_-_s1.zip)

dual extruder ooze prevention set to -30

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1 rc1

### Operating system

windows 10

### Printer model

Ender 5 dual hotend

stonemandave, 2023-08-28:
I would thought that's how ooze prevention works. So you're saying that because I don't use other layers it will always throw a bad number. Why can't it use both if other layers are zero why doesn't it choose the first layer? I guess close it as it's intended by design. I will just avoid that function as it seems kind of useless.

murk-sy, 2023-08-29:
The "default" temperature is the "Other layers" temperature, even if it seems counterintuitive. The reason first layer temperature even exists is to allow better build plate adhesion. There is no realistic scenario where you wouldn't just use the same numbers for both fields.

The feature I think you're looking for is "set extruder temperature to 0 once it is not required in the print anymore".
From my quick testing, Prusaslicer doesn't seem to do that. You should probably make a feature request - if not, I likely will once I get the XL in the office, and have some time with it.

lukasmatena, 2023-09-01:
@stonemandave Now when I read the problem again, I agree that it should be fixed and that the logic should work with 'first layer' temperature in case when 'other layers' temperature is set to zero. We will fix it in the next release after 2.6.1 (which is already too close to introduce new changes).

stonemandave, 2023-09-01:
Thanks it took me a while to figure out what was happening.

lukasmatena, 2023-10-21:
Fixed in [PrusaSlicer 2.7.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-alpha1). Closing.

closed 2023-10-21 by lukasmatena