prusa3d/PrusaSlicer #11196: holes after slicing

donduce, 2023-08-28:
### Description of the bug

holes after slicing on all 4 sides, tested with ver. 2.6 and 2.6.1-rc1, with superslicer 2.5.59.2 there are no holes.
the problem only exists if the duct is sliced while standing on the flange

### Project file & How to reproduce

[image]

[image]

[PrusaSlicer_Mantis nf sunrise v18.zip](https://github.com/prusa3d/PrusaSlicer/files/12451449/PrusaSlicer_Mantis.nf.sunrise.v18.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6 and 2.6.1-rc1

### Operating system

Windows 11

### Printer model

Tronxy X5sa and Ender3

rtyr, 2023-08-28:
You have to either disable "detect thin walls" option or switch perimeter generator to "Arachne".

donduce, 2023-08-28:
thanks, that was the problem

closed 2023-08-28 by donduce