prusa3d/PrusaSlicer #11197: Unsuitable G-code of ramming on Toolchange parameters with multi extruder MM printers ()

monohoshi, 2023-08-28:
### Description of the bug

The new ramming feature for the multi-extruder may generate inappropriate G-code at the wipe tower.
When modeling with two colors as shown in the figure, unnecessary ramming is performed at the end of G-code.
[image]

And, prusaslicer crashes when slicing if the ramming function and the  prime all printing extruders option at wipe tower section are used at the same time.
[image]

### Project file & How to reproduce

See above, please.

[sample.zip](https://github.com/prusa3d/PrusaSlicer/files/12456202/sample.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.6.1-rc1

### Operating system

Windows11

### Printer model

Snapmaker2.0 dual extruders

kubispe1, 2023-08-30:
1) Unnecessary ramming at the wipe tower - This is quite known behaviour
> The new ramming feature for the multi-extruder may generate inappropriate G-code at the wipe tower.
> When modeling with two colors as shown in the figure, unnecessary ramming is performed at the end of G-code.

2) Access violation - seems to me like a bug. I attached your 3mf with **single_extruder_multi_material_priming on** where I can  reproduce the issue (2.6.1-rc1). We will investigate it more. Thanks
[sample_Prime_all_printing_extruders.zip](https://github.com/prusa3d/PrusaSlicer/files/12472741/sample_Prime_all_printing_extruders.zip)
> And, prusaslicer crashes when slicing if the ramming function and the prime all printing extruders option at wipe tower section are used at the same time.

lukasmatena, 2023-09-05:
Fixed in [2.6.1-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.1-rc2). Closing.

closed 2023-09-05 by lukasmatena