prusa3d/PrusaSlicer #11201: Cannot send to printer gcode when there are RESOLVING ADDRESS notifications

Matszwe02, 2023-08-28:
### Description of the bug

When I try to upload some gcode to a printer that isn't connected to the network, then connect it and click again, nothing happens. It appears like it's some sort of queue of resolving address, when the old messages are stuck and prevent to properly resolve address. These messages also don't disappear, I cannot cancel them, only way is to quit PrusaSlicer.
[image]

### Project file & How to reproduce

- Set up printer connection with DNS address (like `printer.local`)
- disconnect printer from network
- click on send gcode to printer
- connect again
- click send gcode again
- see that it's stuck and nothing can help

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 10

### Printer model

Voron 0

Zemistr, 2023-09-12:
Same problem with MK4 and PrusaConnect

seanmcne, 2024-04-06:
This just happened to me now twice in the past couple of days.  I send one or two prints and it works fine, then all the sudden it does what you've pictured above and is stuck in that state until I close/reopen. During this time I can manually export and upload and if I edit my printer settings that also says it tests successfully. 

[image]

after getting those errors I can edit the physical printer and test it, like this: 
[image]

And then when I try again - it still fails: 
[image]

Appears as if it may be related to, or the same issue, as #10944 - though my experience here has been on a windows 10 and windows 11 PC.

mw75, 2024-08-05:
Problem still exists with version 2.8.0-linux-x64-GTK3. Restart is a workaround but still annoying.

closed 2026-09-29 by prusaslicer-bot[bot]