prusa3d/PrusaSlicer #11202: PrusaSlicer creating solid infill incorrectly/weird

GitHubGuy89, 2023-08-29:
### Description of the bug

Not sure if this is a bug or feature request... I'm coming from version 2.4.2 and noticing that the new versions seem to be creating solid infill oddly. Would be nice if there was an option to revert back to the old style of infill. I tried using Classic slicing mode vs Arachne, thinking that would solve it, but it's the same on both.
Also using v2.6.1 RC1 and same issue. Not sure if this is an issue in v2.5 as well.

Attached photos of examples comparing it to 2.4.2.

Problem:
Instead of it continuing the bridging and solid infill all the way inwards, it cuts it off at the 1st perimeter line. For structural objects like this, it's more preferred from a strength standpoint if the infill continued all the way inward like it does in 2.4.2. Also just from a cosmetic standpoint it can create odd patterns sometimes. If there was an advanced option to switch back to the old style of solid infill, depending on what's being printed, I think that would be best.

Also, for the last solid infill layer before the topmost layer, it creates weird triangle patterns instead of a continuous line. This shows through sometimes on the topmost layer, ruining the monotonic top layer pattern. I don't see this as being wanted in any situation.

### Project file & How to reproduce

[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

v2.6.0 and above

### Operating system

Windows 10

### Printer model

Prusa MK3S+

neophyl, 2023-08-29:
Duplicate of several existing issues, included for example this one https://github.com/prusa3d/PrusaSlicer/issues/10573

The new solid infill pattern is mentioned in the release notes for 2.6.

GitHubGuy89, 2023-08-29:
> Duplicate of several existing issues, included for example this one #10573
> 
> The new solid infill pattern is mentioned in the release notes for 2.6.

That does look like the 2nd issue I mentioned. Though it doesn't match up with the 1st issue/request about how it seems to cut off the infill areas when doing bottom/top layers. Unless they are related.

closed 2026-09-29 by prusaslicer-bot[bot]