prusa3d/PrusaSlicer #11208: Arrange function fails to align correctly the objects

antimix, 2023-08-30:
### Description of the bug

The Arrange function do not keep into account skirt & brims and reposition objects too near each other.

In MMU mode,  it now position the objects VERY near the waste tower, and if the object has a brim, it overlaps the brim on the tower.
[image]

In other cases with lots of object, the object near the sheet border have their brim created OUTSIDE the metal plate.
[image]

Clicking again on Arrange after the error causes a new reposition, still failing and raising errors.
[image]

### Project file & How to reproduce

[Arrange_bug.zip](https://github.com/prusa3d/PrusaSlicer/files/12479305/Arrange_bug.zip)

- Load the project
- click on Arrange
- Slice Now
- and it will do a bad positioning and raises errors
- 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1 RC1 Win64

### Operating system

Windows 10

### Printer model

MK3S+ MMU2S

antimix, 2023-09-01:
I discovered that on rounded bed is not capable at all to fit objects correctly!

Here I just added the two main bodies of the MMU3 and aligned them manually:

[image]

Then I clicked on **Arrange** and it immediately it placed an object out of the bed and raised an error !
[image]

Jan-Soustruznik, 2023-09-21:
Hello, @antimix , 

I checked your attached project file. Conflict problem(ERR message) is caused by the generated supports, because objects are too close I saw that you modify the arrange object distance in arrange settings. If you rotate cam under bed you can see that some supports are generated thru model. This behaviour is "normal" because supports are generated during slice of project, so arrange feature doesn't know where the support will be generated.

closed 2026-09-29 by prusaslicer-bot[bot]